cleanflight/cleanflight #2549: SBUS receivers - cannot use RX failsafes

JoeB99, 2016-12-27:
CF failsafe is configured to go into CF stage 1 when the receiver stops sending valid pulses (the receiver is no longer connected to the FC).
This is fine for cPPM.

The CF FS is also set to go into stage 1 when the receiver reports "hey man, i'm in failsafe mode". This occurs when the FS flag is set in an SBUS packet.
Because of this, the RX failsafe CANNOT BE USED with SBUS. 
I believe this is incorrect and stage 1 should only be activated when the receiver is physically disconnected from the FC, and as such, the code that watches the failsafe flag in the sbus packed should be removed, or at the very least there should be a switch to disable this.

The RX failsafe servo position settings should not EVER be ignored.

andre92, 2016-12-27:
Agreed. hold or center on roll/pitch/yaw/throttle is not a valid action for fixed wing.

digitalentity, 2016-12-27:
It will cause inconsist behavior between different receiver types. Safest way is always to report "signal loss" to flight controller and let flight controller decide what to do.

andre92, 2016-12-27:
@digitalentity of course it should be disabled by default and accompanied by warnings to set your RX failsafe, it will only work with digital receivers etc. The easiest way is probably to add preset for roll, pitch, yaw and throttle

digitalentity, 2016-12-27:
@andre92 this issue actually inspired me to open a corresponding one in INAV repo - https://github.com/iNavFlight/inav/issues/978

JoeB99, 2016-12-27:
Would adding a switch to disable "stage 1" fix this for all cases? 

Setting up 16 "SET" servo us positions in CF takes a very long time to do, then I decide to change a "mode" switch position, and forget to change the "SET" us position in CF and boom.. no failsafe.

It's very easy to set failsafe positions with the receiver; set sticks and switches, press a button.

Maybe CF should have a "set failsafe sticks then press this button" on the Stage 1 screen?

digitalentity, 2016-12-27:
@JoeB99 I think second approach is better - it's always safest to make FC aware that radio is lost (or disconnected) - this way failsafe will be consistent between cases when you fly out of range and when your receiver disconnects in flight. A button to "set failsafe from current sticks" is definitely a good thing!

hydra, 2016-12-27:
@ProDrone would like your thoughts and feedback on this please.

closed 2017-02-25 by hydra