cleanflight/cleanflight #2552: CC3D MINI 1.14.0 or 1.14.1 compatiable issue?

lxd811, 2016-12-28:
I recently flash my CC3D to CF 1.14.1&1.14.0(I tried both). I find out when I armed my CC3D by using PPM receiver, my flight controller halts, even in the GUI no new data coming in. I have to disconnect all the 5V wire to get reset and re-connect to GUI. But same thing happens when I arm my CC3D again. 1.13.0 works find for PPM input.
Furthermore, I got two UART1???, one VCP&UART3 in ports menu(FW:1.14.*). is it a bug or something?

andre92, 2016-12-28:
1.14 was released way too early because it was needed to support some new boards. Latest release is 1.14.2-RC3, which should have fixed some issues from 1.14.0 and 1.14.1. If you want to be sure, stay on 1.13.

hydra, 2016-12-28:
@lxd811 try 1.14.3-RC3 and report back on rcgroups forum threads here: 

http://www.rcgroups.com/forums/showthread.php?t=2249574

closed 2016-12-28 by hydra