cleanflight/cleanflight #2558: CC3D Warning LED 6x "the system is overloaded"

ckusnierek, 2016-12-30:
I have a clean flashed CC3D. 
Setup serial Rx SBUS to my Taranis. Inputs are all good. 
Even in the default (reset all) state I get 6 flashes from blue LED indicating a system overload:
[http://shipow.github.io/cleanflight-web/docs/controls.html](http://shipow.github.io/cleanflight-web/docs/controls.html)
ESCs are calibrated and giving ready tones. 
after power on the LED sequence goes something like this:
Fast flash for about 2 seconds
5 flashes (calibrating gyros)
"heart beat" ready flashes for about 3 seconds
the constant 6x flashes indicating a overloaded system
occasionally it will go back to ready pulses for just a few seconds then back to 6x

hydra, 2016-12-31:
try v1.14.2-RC3 please.

closed 2016-12-31 by hydra

ccpyle, 2017-01-04:
Same problem here, even on RC3.