cleanflight/cleanflight #2562: CleanFlight firmware at CC3D

batata004, 2016-12-31:
Hi,

I read many info online about flashing the latest CleanFlight at CC3D (which comes by default with librepilot/openpilot firmware). Some people said I should use GCS (from LibrePilot) to flash the version 1.11 from CleanFlight (which is the latest that comes with .bin file) and after that I should install CleanFlight in my Chrome and try to update to the latest version. Will it work?

I say that cause when I read at Google how to flash CleanFlight there are many misinformation with complicated methods of using an FTDI/Arduino, shorting pins... to flash the latest CleanFlight. Is it really necessary? Can I simply use GCS from LibrePilot to do that and update the firmware using the CleanFlight, will it work?

Engineereric, 2017-01-03:
I've had no success with methods other than using the FTDI USB to serial interface. (It doesn't actually have to be the FTDI it can be any CMOS serial port).
Via the Cleanflight configuration. LEave the baud rate at default.
Also remember once flashed if you want to update the firmware to a newer version you will need to repeat this process so if building CC3D into a frame remember that point.

batata004, 2017-01-03:
@Engineereric thank you for your help :) So as you said, I will give it a try to FTDI USB method. I just would like you to tell me if after I flash firmware+bootloader of Cleanflight, how can I get back to librepilot? Is there someway I flash librepilot back to my cc3d after haved flashed cleanflight?

JoeB99, 2017-01-03:
As a side note, after flashing to clean/betaflight, use an FTDI (or equivalent) into the MAIN port to flash newer versions. In PORTS, MSP needs to be set on for UART1 for this to work.

Engineereric, 2017-01-03:
@batata004, I've never done that but I suspect there is a similar process required to perform the flash. You should perhaps look for answers to this in the Librepilot community.
I should also point out that I thought this discussion board was for bugs and not general set up problems as this is.

closed 2017-02-25 by hydra