cleanflight/cleanflight #2564: How to include new files in makefile?

aliaftab, 2017-01-02:
Hello,

I wrote some new files(drv_adns3080.c and drv_adns3080.h), and in these files I am using the following existing cleanflight files:

**#include "flight/imu.h"
#include "flight/pid.h"**

Now, I want to add the new files in the Makefile. I wrote the following command after "hardware_revision.c " in the "NAZE_SRC" section of the Makefile:

**"drv_adns3080.c "**

But, I keep on getting errors in the "flight/imu.h" and "flight/pid.h" files. If I remove "drv_adns3080.c " line from the makefile, everything runs smoothly i.e. .hex and .efl files are generated.

Could anyone explain it to me how can I add my custom files in the Makefile? Thanks

Regards, Ali

unitware, 2017-01-02:
To get help you need to supply the error messages and the files you have created, at least the makefile.

unitware, 2017-01-02:
When editing makefiles it's vital that you pay attention to the whitespaces (tabs, spaces) and the line continuation character \  that must be last in the line with no whitespaces after.

aliaftab, 2017-01-04:
Hello unitware,

Thanks for your reply. In the following link you can see the project files:
https://www.dropbox.com/s/9zk35pobilmv448/cleanflight.rar?dl=0

The two files which I have added are "drivers/drv_ads3080.h" and "drivers/drv_ads3080.c". My question is how can I include "drivers/drv_ads3080.c" in the "Makefile"?

Regards, Ali

unitware, 2017-01-04:
No problem with the makefile, however as cleanflight is laid out at the moment you need to include lots and lots of header files, not only the ones you need but also the include files that they need (I do not consider this clean, but thats for another discussion).

So, first look at your compiler errors, then search for the correct file and include it.
```
> make TARGET=NAZE 
%% drv_adns3080.c
In file included from ./src/main/drivers/drv_adns3080.c:7:0:
./src/main/flight/imu.h:23:23: error: 'XYZ_AXIS_COUNT' undeclared here (not in a function)
 extern int32_t accSum[XYZ_AXIS_COUNT];
```

Then find the include file that defines XYZ_AXIS_COUNT and include it before it's used:
```
#include "drv_adns3080.h"
#include "common/axis.h"
#include "flight/imu.h"
```

And continue to do so until it builds cleanly.

aliaftab, 2017-01-04:
Hello unitware,

Thanks for your fast reply.
I did like you told me and everything built smoothly. You made my day. Thanks a lot.

Regards, Ali.

unitware, 2017-01-04:
Great, can you close the issue if you are satisfied?

closed 2017-01-04 by aliaftab