cleanflight/cleanflight #2566: SP Racing F3 flight controller doesn't power the PPM Receiver

RustyMcBeefy, 2017-01-02:
When connected to the pdb and the battery, the flight controller doesn't power the ppm radio receiver so I can't bind it to the transmiter nor fly the drone

xxeffects, 2017-02-09:
hello, I have just purchased an Eachine wizard x220 ready to fly. I had to flash betaflight to it (odd considering it is "ready to fly"). After all my settings were setup to my liking (i had only assigned switches and rotated the board 270 degrees), I put everything back together and went to fly it. it would turn on, but wouldnt arm. after testing, i had figured out that i had the exact same problem. This is the only forum addressing this problem. i really need help and really want to fly m y quad! have you figured anything out yet? Need help!!!

rando21, 2017-02-20:
I have this same problem. I used Josh Bardwell settings to get my flight controller working and to come up in Betaflight (his videos are awesome btw) but the transmitter is connected to the receiver but its not controlling the quad. anyone find a solution?

closed 2017-02-25 by hydra