cleanflight/cleanflight #2572: can not arm quad unless plugged into clean flight

buzznlush, 2017-01-10:
hi.
i recently purchased a rotorx rx122 jr and a taranis x9d plus transmitter, everything seemed to set up well. plugged into clean flight with bat plugged into quad and transmitter on after saving and rebooting the quad will arm (switch g) for arming . i can unplug and all stick inputs work well. as soon as i unplug batt and plug back in quad wont arm unless i repeat the steps above.
taranis stick end points are min 1000 max 2000
in cleanflight min throttle is 1050 max is 2000
min command is 1000.
when i do have everything plugged into clean flight and hit save and reboot i notice in the log screen everything works except rebooting comes up in red writing. so perhaps a rebooting problem. 
any help would be much appreciated, ive been stuck here for days

andre92, 2017-01-11:
please post the output of following CLI commands:
dump
status

buzznlush, 2017-01-11:
Hi Andre
Thanks for responding.  I will do what you've asked very soon. I'm beat for
what is happening but it is something to do with the arming switch and
software. it doesn't seem to be hardware related ... well i don't think so
anyways

Darren

buzznlush, 2017-01-11:
[#dump.txt](https://github.com/cleanflight/cleanflight/files/698460/dump.txt)

# dump

# version
# BetaFlight/COLIBRI_RACE 3.0.0 Oct  6 2016 / 09:00:01 (f408ed9)

# name
name -

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature VBAT
feature INFLIGHT_ACC_CAL
feature RX_SERIAL
feature FAILSAFE
feature LED_STRIP
feature OSD
feature AIRMODE

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
[... 353 lines cut ...]
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 70
set rc_rate_yaw = 80
set rc_expo = 15
set rc_yaw_expo = 15
set thr_mid = 50
set thr_expo = 0
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
set tpa_rate = 20
set tpa_breakpoint = 1650

# status
System Uptime: 24 seconds, Voltage: 0 * 0.1V (1S battery - NOT PRESENT), CPU:26%
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000
Cycle Time: 135, I2C Errors: 1, config size: 1476

#

buzznlush, 2017-01-11:
done, 
i have changed to betaflight and the fc now has betaflight 3.0.0 aton v2 but the issue remains the same

andre92, 2017-01-11:
If you still have the same issue with Betaflight, could you please also post it [in the Betaflight issue tracker](https://github.com/betaflight/betaflight/issues)?

buzznlush, 2017-01-11:
the dump I posted was from bf. .. i swapped to bf because I thought cf may
have been the issue.
I flashed the fc with beta flight atom v2 and had the same issue..

Darren

buzznlush, 2017-01-11:
Oh. . yes I can do that .. i get what your saying

Darren

waynewendrone, 2017-02-09:
I have the same problem here and don't know what to do, I'm using an sp racing f3

buzznlush, 2017-02-09:
Hi Wayne.
the problem I had (rotorx arom v2 Tbs power cube ))was solved by going
through the osd setup .. at the time I had no goggles and didn't read
anywhere I had to go through the osd set up even though now looking at
things it made sense to do so.
other than that i discovered it can also be minimum throttle settings
initially I had a min throttle of 1000 i had to reduce it to 989 before the
quad would arm (srd280 v3 cc3d fc )

Darren

waynewendrone, 2017-02-09:
Ok that's great that we have a possible solution, could you just give me the exact things I need to look into and the steps I need to take. I'm new to drones:D!

Wayne

waynewendrone, 2017-02-09:
About the minimum throttle, surely the quad wouldn't have armed while during clean flight of the numbers were too high?

buzznlush, 2017-02-10:
True. I'd day you have everything right but have not completed something.

With the power cube i had to press a button on the vtx and follow the
prompts after which it armed armed and flew without issue.
I'm new to this myself so I may not be as much help as some of the other
more experienced pilots.

Darren

buzznlush, 2017-02-10:
when you say it armed when plugged into cleanflight was that using the tx
or when testing motor outputs in a screen

Darren

waynewendrone, 2017-02-10:
What I mean is that I can arm it using my Devo 10

waynewendrone, 2017-02-10:
But can't while USB is unplugged

buzznlush, 2017-02-10:
do you have a failsafe set on the fc . I had that mentioned to me before I
sorted the issue I was having.
As I said.  The issue I had was specific to the vtx board which is part of
the Tbs powercube . It had to be calibrated via osd before it would arm
without being plugged in to pc

Darren

waynewendrone, 2017-02-11:
I'm not getting any beeps from the esc without the USB cable plugged in, but if I have the USB cable plugged in first and then the battery the motors would arm

waynewendrone, 2017-02-11:
here is my dump

waynewendrone, 2017-02-11:
dump

# version
# Cleanflight/SPRACINGF3 1.14.2 Dec 31 2016 / 02:11:08 (747971f)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -AMPERAGE_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -OSD
feature VBAT
feature MOTOR_STOP
feature FAILSAFE
feature RX_PARALLEL_PWM

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# led
led 0 15,15:ES:AI:0
led 1 15,8:E:FW:0
led 2 15,7:E:FW:0
led 3 15,0:NE:AI:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:AI:0
led 7 0,7:W:FW:0
led 8 0,8:W:FW:0
led 9 0,15:SW:AI:0
led 10 7,15:S:FW:0
led 11 8,15:S:FW:0
led 12 7,7:U:FW:0
led 13 8,7:U:FW:0
led 14 7,8:D:FW:0
led 15 8,8:D:FW:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
[... 314 lines cut ...]
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

buzznlush, 2017-02-12:
Hi Wayne.
just wondering how you are doing with the issue at hand.
I had a look at the dump file but because I'm almost as new ass you I'm
can't see much.
I can only think of throttle settings, failsafe rx tx and fc and osd setup.
. Each of those if not set will cause your issue.

Darren

waynewendrone, 2017-02-13:
Hi Darren, it's the same as it's always been, I can arm the motors if I plug the USB into the fc and then into a power source, I was wondering if it could be that my fc is not getting the 5v it needs but I have linear Bec in my esc so I don't see how that's the case, also I'm sure how to configure the osd since I don't have one.

Wayne

hydra, 2017-02-14:
please use rcgroups for user-support

closed 2017-02-14 by hydra

jalao, 2018-02-15:
All, has anyone found a solution to BFF4 not arming unless plugged into to USB. I purchased two FC which cost me almost $90 including shipping. I purchased this FC because I trusted the teams developing the BF can be trusted. It is very frustrating after 2 separate builds  with 2 FC's and I can not arm my quad. I have increased my Moron_threshold to 120. Nothing happens. Please if there is a solution please kindly help.

mikeller, 2018-02-15:
@jalao: This seems to be a problem with the board design, caused by the voltage regulator introducing too much noise for the gyro to work when not plugged into USB. Contact the manufacturer.

waynewendrone, 2018-02-15:
What about turning off the accelerometer and fly in rate mode??