cleanflight/cleanflight #2576: no gps on softserial1 with 1.14.2 @ CC3D

tita1705, 2017-01-15:
Although softserial is configured the right way, GPS is not regocnized. The command 'serialpassthrough 30 9600' is working and showing the serial gps dump. Connected is a common NMEA gps module.

Same config worked with 1.13.0 without problems.

SERIAL PASSTHROUGH DUMP:

# serialpassthrougPort 30 opened, baud=9600
Relaying data to device on port 0, Reset your board to exit serial passthrough mode.
$GPGGA,131616.166,,,,,0,2,,,M,,M,,*49
$GPGLL,,,,,131616.166,V,N*79
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,3,1,11,30,74,174,21,05,60,286,,07,58,064,,09,30,091,27*7A
$GPGSV,3,2,11,13,25,274,,02,20,233,,28,16,153,,20,12,314,*73
$GPGSV,3,3,11,06,02,199,,16,01,018,,27,01,043,*40
$GPRMC,131616.166,V,,,,,0.00,237.69,150117,,,N*44
$GPVTG,237.69,T,,M,0.00,N,0.01,K,N*3A
$GPGGA,131617.166,,,,,0,2,,,M,,M,,*48
$GPGLL,,,,,131617.166,V,N*78
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,3,1,11,30,74,174,22,05,60,286,,07,58,064,,09,30,091,27*79
$GPGSV,3,2,11,13,25,274,,02,20,233,,28,16,153,,20,12,314,*73
$GPGSV,3,3,11,06,02,199,,16,01,018,,27,01,043,*40
$GPRMC,131617.166,V,,,,,0.02,111.07,150117,,,N*48
$GPVTG,111.07,T,,M,0.02,N,0.03,K,N*35
$GPGGA,131618.166,,,,,0,2,,,M,,M,,*47
$GPGLL,,,,,131618.166,V,N*77
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,3,1,11,30,74,173,23,05,60,286,19,07,58,064,,09,30,091,27*77
$GPGSV,3,2,11,13,25,274,,02,20,233,,28,16,153,,20,12,314,*73
$GPGSV,3,3,11,06,02,199,,16,01,018,,27,01,043,*40
$GPRMC,131618.166,V,,,,,0.06,76.98,150117,,,N*75
$GPVTG,76.98,T,,M,0.06,N,0.12,K,N*07
$GPGGA,131619.166,,,,,0,2,,,M,,M,,*46
$GPGLL,,,,,131619.166,V,N*76
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,3,1,11,30,74,173,23,05,60,286,,07,58,064,,09,30,091,26*7E
$GPGSV,3,2,11,13,25,274,,02,20,233,,28,16,153,,20,12,314,*73
$GPGSV,3,3,11,06,02,199,,16,01,018,,27,01,043,*40
$GPRMC,131619.166,V,,,,,0.07,65.67,150117,,,N*77
$GPVTG,65.67,T,,M,0.07,N,0.14,K,N*02
$GPGGA,131620.166,,,,,0,2,,,M,,M,,*4C
$GPGLL,,,,,131620.166,V,N*7C
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GPGSV,3,1,11,30,74,173,23,05,60,286,,07,5

SETTINGS DUMP:

# version
# Cleanflight/CC3D 1.14.2 Dec 31 2016 / 02:12:54 (747971f)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -AMPERAGE_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -OSD
feature RX_PPM
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature FAILSAFE
feature DISPLAY
feature BLACKBOX

# map
map AETR1234

# serial
serial 20 1 115200 9600 0 115200
serial 0 1 115200 9600 0 115200
serial 2 0 115200 9600 0 115200
serial 30 2 115200 9600 0 115200

# led
led 0 15,15:ES:AI:0
led 1 15,8:E:FW:0
led 2 15,7:E:FW:0
led 3 15,0:NE:AI:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:AI:0
led 7 0,7:W:FW:0
led 8 0,8:W:FW:0
led 9 0,15:SW:AI:0
led 10 7,15:S:FW:0
led 11 8,15:S:FW:0
led 12 7,7:U:FW:0
led 13 8,7:U:FW:0
led 14 7,8:D:FW:0
led 15 8,8:D:FW:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0::C:0
led 29 0,0::C:0
led 30 0,0::C:0
led 31 0,0::C:0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

# mode_color
mode_color 0 0 1
mode_color 0 1 11
mode_color 0 2 2
mode_color 0 3 13
mode_color 0 4 10
mode_color 0 5 3
mode_color 1 0 5
mode_color 1 1 11
mode_color 1 2 3
mode_color 1 3 13
mode_color 1 4 10
mode_color 1 5 3
mode_color 2 0 10
mode_color 2 1 11
mode_color 2 2 4
mode_color 2 3 13
mode_color 2 4 10
mode_color 2 5 3
mode_color 3 0 8
mode_color 3 1 11
mode_color 3 2 4
mode_color 3 3 13
mode_color 3 4 10
mode_color 3 5 3
mode_color 4 0 7
mode_color 4 1 11
mode_color 4 2 3
mode_color 4 3 13
mode_color 4 4 10
mode_color 4 5 3
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3
mode_color 6 0 6
mode_color 6 1 10
mode_color 6 2 1
mode_color 6 3 0
mode_color 6 4 0
mode_color 6 5 2
mode_color 6 6 3
mode_color 6 7 6
mode_color 6 8 0
mode_color 6 9 0
mode_color 6 10 0

set debug_mode = NONE
set emf_avoidance = OFF
set i2c_highspeed = ON
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set rc_smoothing = OFF
set rx_min_usec = 885
set rx_max_usec = 2115
set serialrx_provider = SPEK1024
set sbus_inversion = ON
set spektrum_sat_bind = 0
set input_filtering_mode = OFF
set min_throttle = 1080
set max_throttle = 2000
set min_command = 1000
set motor_pwm_rate = 400
set servo_center_pulse = 1500
set servo_pwm_rate = 50
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set retarded_arm = OFF
set disarm_kill_switch = ON
set auto_disarm_delay = 5
set max_arm_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set telemetry_switch = OFF
set telemetry_inversion = OFF
set telemetry_send_cells = ON
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set amperage_meter_scale = 0
set amperage_meter_offset = 0
set multiwii_amperage_meter_output = OFF
set amperage_meter_source = VIRTUAL
set vbat_hysteresis = 1
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set small_angle = 25
set max_angle_inclination = 500
set pid_process_denom = 1
set gyro_sync = 1
set gyro_sample_hz = 2000
set gyro_lpf = OFF
set gyro_lowpass_level = NORMAL
set gyro_lowpass_hz = 90
set gyro_notch_hz = 130
set gyro_notch_cutoff_hz = 130
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set pid_at_min_throttle = ON
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq =  400.000
set servo_lowpass_enable = OFF
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = 0
set acc_hardware = 0
set baro_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 5
set blackbox_device = SPIFLASH
set magzero_x = -188
set magzero_y = 33
set magzero_z = -195

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

dkjab, 2017-02-08:
Confirmed. Switching temporarily from SoftSerial1 to UART1, GPS works fine.
The same issue exists in other forks (eg. INav).  
NMEA and UBLOX modes both affected. Both ports  tested at 19200 baud 5hz.

dkjab, 2017-02-08:
SoftSerial is commented out in target/CC3D/target.h.

//#define USE_SOFTSERIAL1

dkjab, 2017-02-08:
GPS on CC3D disabled by Hydra. See below:

https://github.com/cleanflight/cleanflight/commit/5b09fe605be3aa753c11198e659eea32cc47caa8

Surely removing display support (as per NAZE32) would be a better code reduction solution?

closed 2017-02-25 by hydra