cleanflight/cleanflight #2578: Failsafe/disarm not functional when crashed upside down in 1.14.2

Engineereric, 2017-01-17:
I need to do more tests but have seen this now on 3 different boards in 3 quads.
1 CC3D and two Naze32(one acro the other 10DOF)
When crashed and upside down motors keep spinning regardless of RC input or disarm function.
Once you kick it right side up it stops.
Today's crash ripped the radio from the quad and it lay there upside down whipping up dust.
Had to give it a kick to right it and then it stopped and beeped.
I will see what I can do to replicate the problem on the bench after work.

andre92, 2017-01-18:
I suspect your failsafe is not setup as it should. Set the failsafe procedure to DROP in the failsafe tab.
That should get it disarmed when receiver gets ejected.
Also, set your receiver to no pulses when you lose radio contact. For most receivers you can do that by pressing the button for a second when the transmitter is off.

Engineereric, 2017-01-20:
No I dont think so. 
Fail safe primary is set with all controls to Auto and arm switch to disarm level.
Stage 2 fail safe is set to activate after 1 second and is set to drop for 5 seconds.
The receiver was lost in only one crash and the other crashes it should still have had good reception. 
The disarm function could not be activated and it took well over 30 seconds to get to the crash.
During which at least one motor was still spinning.
I am doing a rebuild now as wires were pulled form the FC and when it is back together I will attempt to replicate problem physically with no props attached.
Thanks
Eric.

andre92, 2017-01-20:
If you arm the quad, throttle up a bit, then turn off your transmitter, what happens? Does it disarm or does it keep going until you turn on your transmitter and regain control?

Engineereric, 2017-01-20:
It stops dead. I've had it set that way for a long time because when I first started flying I had an antenna break and a quad took off across the road at over 7G until it hit a wall. Took less than 2 seconds to travel 70m! That's because it kept the last valid level that mus have been just in the range of validity for MAX throttle.
Also the local club requires us to perform that test on their fly days before we are allowed to fly.
I cant vouch for what my friend has set up on he's.

andre92, 2017-01-20:
What beeped when you kicked the quad straight up, the ESCs or the onboard buzzer? Maybe your flight controller was rebooting.
Are you using oneshot125 or unsynced pwm?

Engineereric, 2017-01-20:
motors stopped and it was safe to pick up. no reboot or dissarm beeps that I can recall.

closed 2017-02-25 by hydra