cleanflight/cleanflight #2579: 1.14.2 Not Arming or Arming Sporadically

imtjuanjose, 2017-01-17:
I had the problem of the repeated UART1 on v1.14.1 but it always ARMED right.

After I installed 1.14.2:

I cannot ARM the quad when the Jaw in Board and Sensor Alignment is set to -90 or 270 Degrees (phisically rotated the Board due to easier conection of ports.

When the Jaw in Board and Sensor Alignment is set to 0 Degrees it ARMs Sometimes (one in each 7 times maybe).

In a previous flash of 1.14.2 configurated the LED Strip for Warnings and they showed the Blinking Green Color (Which I think means ARM BLOCKED).

CC3D Evo (Same CC3D but with a Satellite connector))
TARANIS -plus Transmitter
FrSKY X4R Receiver (SBUS - 1234TAER)

I already tried calibrating, using AUX Channel to ARM, increasing the max_arm_angle, small_angle and many other things I have flashed many times both versions and the same happens, as I see it with the same config works fine in 1.14.1 and is not working in 1.14.2, due to the fact that the Board Alignment affects the ARMING i think that is here where the problem is, also I saw just a momento ago someone posting that the drone didn't disarm while flipped.

I will post the DUMP of both 1.14.1 and 1.14.2:

#Cleanflight CC3D 1.14.2 (NOT ARMING)

With Board and Sensor Alignment (-90 Yaw Degrees) - Never ARM.

With Board and Sensor Alignment (0 Yaw Degrees) - Sometimes ARM.

Flashed using UART TTL MicroUSB and Flash Loader Demonstrator.

> Ports
> *UART1 (Serial)
> 
> Configuration
> *Board and Sensor Alignment (-90 Yaw Degrees)
> *Blackbox (OFF)
> 
> Receiver
> *Receiver Mode (RX_SERIAL)
> *Serial Receiver Provider (SBUS)
> 
> CLI
> *map 1234TAER

```

> # dump
> 
> # version
> # Cleanflight/CC3D 1.14.2 Dec 31 2016 / 02:12:54 (747971f)
> # dump master
> 
> # mixer
> mixer QUADX
> mmix reset
> smix reset
> 
> 
> # feature
> feature -RX_PPM
> feature -VBAT
> feature -INFLIGHT_ACC_CAL
> feature -RX_SERIAL
> feature -MOTOR_STOP
> feature -SERVO_TILT
> feature -SOFTSERIAL
> feature -GPS
> feature -FAILSAFE
> feature -SONAR
> feature -TELEMETRY
> feature -AMPERAGE_METER
> feature -3D
> feature -RX_PARALLEL_PWM
> feature -RX_MSP
[... 752 lines cut ...]
> set horizon_tilt_mode = SAFE
> 
> # dump rates
> 
> # rateprofile
> rateprofile 0
> 
> set rc_rate = 90
> set rc_expo = 65
> set rc_yaw_expo = 0
> set thr_mid = 50
> set thr_expo = 0
> set roll_rate = 0
> set pitch_rate = 0
> set yaw_rate = 0
> set tpa_rate = 0
> set tpa_breakpoint = 1500
```

I will like to Thank You for the great job you do with this awesome project and thanks in advance to anyone that can help me with this issue.

PD: I have a video of the testing process of both versions in case is needed, but you tell me how can I assist.

Grettings.
Juan Jose Guichard Romero

andre92, 2017-01-18:
Correct, blinking led indicates arm blocked. I suspect a system load of 2.31 is slightly too high. Try getting it to something like 0.231. Your cpu is way overloaded. Could you post the output of "tasks" as well?

imtjuanjose, 2017-01-20:
@andre92 Thank you so much! Changing the gyro_sample_hz from 2000 to 1000 solved the problem, now is arming 100% of times. 

Here is the Task´s output:

```
> # tasks
> Task list          max/us  avg/us rate/hz maxload avgload     total/ms
>  0 -       SYSTEM       4       0       9    0.5%    0.5%         0
>  1 -         GYRO     271     177     996   27.4%   18.1%      2848
>  2 -          PID     247     201     991   24.9%   20.4%      3086
>  3 -        ACCEL     207     158     988   20.9%   16.1%      2613
>  4 -     ATTITUDE     313     250      98    3.5%    2.9%       389
>  5 -           RX     143     112     111    2.0%    1.7%       217
>  6 -       SERIAL     725       2      99    7.6%    0.5%        51
>  8 -  HW_WATCHDOG       1       0       0    0.5%    0.5%         0
>  9 -       BEEPER       7       0      98    0.5%    0.5%         2
```

# 

If it's not too much to ask, ¿What other changes may I do to reduce the system load?

The drone is for a school project, I do not require it to fly soo fast, do 3D flips or anything hardcore like that, in fact it will fly really slow. But I would like to use LED_Strips, SBUS & Telemetry with the TARANIS and if I can get away with it maybe an OSD and I2C OLED Screen or even a Sonar.  (I'm sorry if this is not the place for such questions as this is for throubleshooting of the Firmware versions, but I thing that maybe it would be useful for someone else with the same problem that reads this).

Thank you for the advice with the small_angle and for all the help, I really appreciate it.

PD: This is the new Status OUTPUT:
# status
System Uptime: 796 seconds, Voltage: 0 * 0.1V (3S battery - OK), System load: 0.12
CPU Clock=72MHz, GYRO=MPU6000, ACC=MPU6000
PIDd: 1098, GYROd: 1099, I2C Errors: 0, registry size: 48

Greetings
Juan Jose

andre92, 2017-01-20:
Glad it works now :)
System load of 0.12 is very respectable, all tasks look good. If you want smartport telemetry you're going to need softserial, which is a rather heavy feature. I'm also not sure it can be run at the same time as Sonar or Led_strip. OSD should fit on a Uart, and serial receiver (sbus) is supported on CC3D. Please refer to a manual for the board, as here: https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20CC3D.md

imtjuanjose, 2017-01-20:
I will do so. Thank you for all the information and help.

Have a nice day.

Greetings
Juan Jose

closed 2017-01-20 by imtjuanjose