cleanflight/cleanflight #2586: feature request: wireless USB simulator

R5fan, 2017-01-20:
I use my taranis with PC simulators (FPV event, hotprops etc). Which works well, but requires a clunky USB cable. I want to cut the cord. I bought a USB adapter that connects to a receiver:
https://hobbyking.com/en_us/wireless-9-in1-simulator-adapter.html

Which is perfect in theory, but turns out almost useless in reality, as the input becomes very unprecise, jittery and with significant latency. I cant find another product that does the same, expect for Spektrum.

That made me wonder, if we couldnt use a spare flight controller board to take any SBUS/PPM receiver, and make it act as a (accurate, low latency) USB gaming device. This may not strictly be cleanflight functionality, but it would be awesome to have, so Im posting it here.

marcroe, 2017-01-20:
Check out https://github.com/marcroe/uinputmw which is something I hacked together in a day or so and basically does what you describe, but only under Linux. It works fine with all the simulators that run under Linux. It requires python and the uinput python package to run.
Please remove props or power the receiver through USB, should you want to test this with an FC part of an assembled copter instead of a spare FC and RX ;-)

unitware, 2017-01-20:
"This may not strictly be cleanflight functionality, but it would be awesome to have, so Im posting it here."
I second that, in both ways. Now close this issue and help close a few others as a good gesture so we can get the huge heap of issues in a more manageable state :)

johnty, 2017-01-27:
apologies for randomly hijacking this thread, but after playing around a bit with cleanflight/beta flight, testing the RX through USB, and messing around with [smartpropoplus ](http://www.smartpropoplus.com/site/) i've kinda reached a similar conclusion that it would be neat for beta/cleanflight to support this feature through the FC boards, even if its a bit of a tangent in terms of features

a really neat standalone way would be to use HID emulation (if the microcontroller supports it), which could directly instantiate a joystick device and provide virtually driver-free joystick capability. however this would require additional code for the firmware, and potentially be problematic when using the USB serial port for regular configuration. perhaps a better way would be to get the current "configurator" programs to create the devices, since they already read/display the RX data anyway in real time. there's a lot of existing javascript libraries for _accessing_ devices on your computer, but i don't know of any that will create a virtual device. i think some lower level programming is required on that part... but it would be a neat project!

just tossing around some ideas... ;)

hydra, 2017-02-05:
yes we could do this, no not in cleanflight, however feel free to fork the codebase and create something that works.  maybe if the solution is good we can put the project in the cleanflight organization and host the binaries there.

closed 2017-02-05 by hydra