cleanflight/cleanflight #2591: Naze32 rev 5 with CF 1.14.2: wrong accelerometer z value

DrUlf, 2017-01-24:
On a Naze32 rev 5 board, the z value of the accelerometer is wrong.
Please compare the 2 sensor output screen shots. I turned the board through y, x and z axes during that time.
In the screen shot with CF 1.13.0 you see y axis taking value 1 and -1, then x axis with values -1 and 1, and at last z axis with values 1, -1, 1, -1.
In the screen shot with CF 1.14.2 you see y and x axes behaving the same but z axis only moving between 0.8 and 0.2.
[image]
[image]

andre92, 2017-01-25:
Have you tried reflashing and calibrating? I have been unable to replicate this behaviour

DrUlf, 2017-01-25:
Yes, the screen shots are taken after a reflash with full chip erase and fresh calibration of the accelerometer.

andre92, 2017-01-25:
do you have a CLI dump? It looks like something is scaling the Z axis to 50%

DrUlf, 2017-01-26:
Please find attached dump and screen shot of the following scenario:
- Flash CF 1.14.2 with full chip erase
- Connect
- Calibrate accelerometer
- Dump
- Take sensors screen shot (again I rotated through all axes).

[image]

2017-01-26 @ 11:14:50 -- MultiWii API version received - 1.25.0
2017-01-26 @ 11:14:50 -- Flight controller info, identifier: CLFL, version: 1.14.2
2017-01-26 @ 11:14:50 -- Running firmware released on: Dec 31 2016 02:12:34
2017-01-26 @ 11:14:50 -- Board: AFNA, version: 2
2017-01-26 @ 11:14:58 -- Accelerometer calibration started
2017-01-26 @ 11:15:00 -- Accelerometer calibration finished
2017-01-26 @ 11:15:03 -- CLI mode detected
Setup
Ports
Configuration
Failsafe
PID Tuning
Receiver
Modes
Adjustments
Servos
GPS
Power & Battery
Motors
Race Transponder
LED Strip
Sensors
Tethered Logging
Blackbox
CLI
Note: Leaving CLI tab or pressing Disconnect will automatically send "exit" to the board. With the latest firmware this will make the controller restart and unsaved changes will be lost.

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# Cleanflight/NAZE 1.14.2 Dec 31 2016 / 02:12:34 (747971f)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -AMPERAGE_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
[... 356 lines cut ...]
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

#

DrUlf, 2017-01-26:
I have just pressed "Reset settings" on "Setup" page. The z value of the accelerometer is now constantly at -4.6.

[image]

The Setup page now shows this picture as if the quad is upside down.

[image]

DrUlf, 2017-01-26:
Its me again.
I retested all CF versions from 1.11 to 1.14.2. Now all versions consistently show the strange behavior.
Is this a proof for a board issue? It is a defect of the accelerometer?

andre92, 2017-01-26:
even an uncalibrated accelerometer should not show as much as 4.6G on any axis in any direction. I am afraid there is a hardware issue.

hydra, 2017-02-05:
yes, hardware issue.  nothing we can do here, closing.

closed 2017-02-05 by hydra