cleanflight/cleanflight #2594: SPRacingF3OSD with SPRacingF3EVO - Video and OSD Overlay issues

JBlocks, 2017-01-25:
In use is a SPRacingF3OSD and a SPRacingF3EVO with the "Stackable Design". It is the latest CleanFlight version 1.14.2 installed on the SPRacingF3OSD and on the SPRacingF3EVO as well.
Stackable design, see: http://seriouslypro.com/img/spracingf3osd/misc/pcbs.jpg

When the PDB / OSD is connected to the battery, the Cleanflight logo and the video overlay (OSD) will not be displayed. After I have recharged the battery one, two, three, four, ... times, the Cleanflight logo is displayed in the video overlay and also the voltage, mAh, Power-On Time, etc is shown.

I thought there is a problem with the current CleanFlight Version 1.14.2, but the previous versions have the same problem. I have tried it with the previous CleanFlight version 1.14.1. Without success!

On start (connecting battery to the F3OSD), there are several video problems and the synchronization of the video image. 30 seconds and up to 10 minutes. See the appendix of this ticket.

The camera (RunCam Eagle) is connected to the CAM connector and the video transmitter (TBS Unify HV Race) is connected to the VTX via the JST-Z connectors.
The camera and the video transmitter are each supplied with 12 V of voltage. (Bridged via soldering on the pins)

If the camera and the video transmitter are connected directly (without the ODB), I get a videosignal without disturbances and of course without OSD. The problem is not the camera or the video transmitter.

How to solve this problem? - I would not have to connect the battery several times until I get lucky with a clean video with OSD display.

Attachments:
[Video-with-sync-problems.zip](https://github.com/cleanflight/cleanflight/files/731128/Video-with-sync-problems.zip)
[image]
[image]

hydra, 2017-02-05:
seems like a hardware compatibility issue between the max7456 and the camera.  i doubt there's much anyone can do about that but feel free to check the max7456 datasheet to see if there's any commands we can send to the max7456 that might help.  you can try updating the max7456 driver code to test changes yourself.  If you find anything that helps please create a new PR with the changes required along with any additional information and a link to this issue.

closed 2017-02-05 by hydra

JBlocks, 2017-02-05:
Thanks for your reply. Some topics in the web talking about an issue with cams with 800 TVL. 
I can not fix the problem, because I'm not able to develop or change the source code. I hope you or other developers can help!

- http://en.playuav.com/question/21
- https://www.rcgroups.com/forums/showthread.php?2152696-The-new-RunCam-not-compatible-with-MinimOSD-maybe-others

What I figued out is:
Changing the video mode from PAL/50Hz to NTSC/60Hz in the camera menu and set the video mode in cleanflight to ntsc will boot up the OSD but without any characters or cleanflight logo.
There might be a problem with PAL mode and also the 800TVL! 

Another very strange problem is that the OSD overlay appear and all video problems are gone as soon as I connect the USB cable (USB 2.0) to the ODB. The battery and the USB cable are connected to the powerboard (ODB) and everything works. If I disconnect the USB cable, the video signal breaks and we have the problem as described above.

After 3-5 minutes the issue comes again and I have to disconnect the battery and connect again. 
I'm not able to fly with the SPRacingF3OSD. I am very disappointed.