cleanflight/cleanflight #2595: Servo-tlt in version 1.14.2 does work different of documentation

Hennyr, 2017-01-26:
I have SPRacing F3EVO and Cleanflight version 1.14.2.
After activation Servotilt the first 2 servoconnectors should be for the Gimbalservo's.

This not right in this version. position 5 & 6 are the working connectors.
BUT for manual activation or management of the servo's you should select the used receiverport on servo 0 & 1!!
This is working well, but not in a logical way.
I suggest to correct also the servo-setup; there use servo 5 & 6 according connector use.

hydra, 2017-02-05:
i cannot find the location in the documentation where it says you should use pins 1 and 2 for the servos on the EVO.  can you remind me where you saw it?  please provide a link to the line in the documentation on github.

Hennyr, 2017-02-06:
Dominic,

 

After your question i served the internet and find out that the examples of the servotilt all were withe the naze32-boards.

I have have know also find examples of expiriences with spracing (not with EVO) and they are talking about other pin usage.

 

Sorry for the misleading. But perhaps you can mention in the documentation of SPracing-boards that they pins fot the servo-tilt are different to the most showed video’s and discussions.

 

Thanks

 

 

Met vriendelijke groet,

 

Henny Ribbink

closed 2017-02-25 by hydra