cleanflight/cleanflight #2603: MSP_SET_RAW_RC: Yaw and throttle channels swapped

christianrauch, 2017-02-01:
The original order of channels in the `MSP_SET_RAW_RC` message is defined as:

1. ROLL
2. PITCH
3. YAW 
4. THROTTLE
5. ... 8. AUX1 ... AUX4

(http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol)

According to this order, I should be able to arm via sending the array:
[1500, 1500, **2000**, **1000**, 1000, 1000, 1000, 1000]
This does work for MultiWii 2.4 but not for cleanflight 1.14.2

However, I can only arm cleanflight by swapping the yaw and throttle channels:
[1500, 1500, **1000**, **2000**, 1000, 1000, 1000, 1000]

hydra, 2017-02-05:
likely the RC mapping you're using is wrong, did you check the CF code?

hydra, 2017-02-05:
the default map in CF comes from this line of code:

`parseRcChannels("AETR1234", rxConfig());`

see `config.c`

use MSP_SET_RX_MAP once to set the correct mapping before using MSP_SET_RAW_RC and you'll be fine.

closed 2017-02-05 by hydra

christianrauch, 2017-02-05:
I am referring to the original definition of `MSP_SET_RAW_RC` at http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol. The order of channels is fixed in this definition and is independent from the physical remote control mode.

Shouldn't the MSP implementation in cleanflight follow this definition?

hydra, 2017-02-05:
it does, but then due to the default mapping the channel order is changed.  msp rx is handled like any other rx protocol in CF and so the mapping applies in CF.

christianrauch, 2017-02-05:
Sorry, I disagree. Compared to other RX protocols, the order of channels in `MSP_SET_RAW_RC` is very well defined.
I actually think that the channel mapping should be done in CF to assure that channels in `MSP_SET_RAW_RC` are interpreted in the defined order. Otherwise all clients need to figure out the correct mapping by themselves.

E.g. see the wrong values send by the GUI control sticks. Both images show the left stick in the lower right corner (throttle 1000, yaw 2000) which should arm the Quad. Because of the wrong mapping (`AETR1234` as well `TAER1234`) it won't arm:
[image] 
[image]

This issue could be fixed by interpreting the MSP channels in the defined order.

hydra, 2017-02-05:
no, then there would be two mappings.  

MSP RX is handled like all other RX protocols.  All RX protocols allow the TRANSMITTER to define the channel order and the FC to map the channels into the appropriate functions via the RX MAP.

We do not need nor want additional code or logic to do this when clients can use the API provided.

hydra, 2017-02-05:
see also https://github.com/cleanflight/cleanflight/issues/2359#issuecomment-241808977