cleanflight/cleanflight #2605: No throttle command, Max throttle and only Max from transmitter

astroxboy, 2017-02-04:
Hi All,

So i have recently flashed betaflight on my naze32. I have had this board running cleanflight for some time now with no issues. Setup has been straight forward but i cannot solve this issue that any throttle input on my transmitter causes my motors to instantly spool to max throttle. (not the same as the airmode gradual spool. it is instant!)

Notes:
Calibrated my esc's 4 or 5 times to ensure its not them. 
using a D4r-ii so made sure my receiver is in PPM
checked all my stick commands and they correspond correctly in betaflight
all pids and other reciever input settings are all default.
Airmode is off as originally when i had this on my motors would do the good old start slow and continue to ramp up which i believe is normal. 

When i test my motors in the software page they work fine, i only have this issue when i put in a throttle command via my transmitter. 

If anyone has any idea of what i could be missing (which i probably am) your input would be much appreciated.

andre92, 2017-02-04:
post cli dump please?
I will try to replicate it, have naze and d4r laying around.

hydra, 2017-02-05:
please use rcgroups forums for support first, if you find the issue is in CF then please report it here.

closed 2017-02-05 by hydra