cleanflight/cleanflight #2606: Wrong ids for modes in MSP_BOXIDS

christianrauch, 2017-02-04:
I am trying to determine the active modes via MSP.

I receive the following data:

| index | `MSP_BOXNAMES`  | `MSP_BOXIDS` |
| ------------- | ------------- | ------------- |
| 0	| ARM |	0 |
| 1	| ANGLE |	1 |
| 2	| HORIZON |	2 |
| 3	| BARO |	3 |
| 4	| MAG |	5 |
| 5	| HEADFREE |	6 |
| 6	| HEADADJ |	7 |
| 7	| BEEPER	 |13 |
| 8	| OSD SW	 |19 |
| 9	| GTUNE	 |21 |
| 10	| BLACKBOX	 |26 |
| 11	| FAILSAFE	 |27 |
| 12	| AIR MODE	 |28 |

From  `MSP_STATUS` I receive the mask 2050, which resolves to  `1` and `11`. In the original MultiWii implementation and MSP definition, these numbers are looked up in `MSP_BOXIDS` and mapped to box IDs `1` and `27`. However, there is no `MSP_BOXNAMES` at index `27`.

At this point, I know from the GUI that failsafe is active and it seems obvious that the mask reports the index of the mode in `MSP_BOXNAMES` instead of the index in `MSP_BOXIDS`. E.g. the box indices are not mapped by `MSP_BOXIDS`.

What is the reason for changing the original behaviour?
What do the IDs in `MSP_BOXIDS` represent, if they aren't used for determining the active modes?

ledvinap, 2017-02-04:
@christianrauch : 
there are two MSP commands to return BOX information - MSP_BOXNAMES and MSP_BOXIDS.
Both return only boxes that are active (defined in `initActiveBoxIds`).
MSP_STATUS return bitmask that is indexed same way as these replies - first column in your table.

So when you get bit 11, you get it's name as 11th string in MSP_BOXNAMES (FAILSAFE) and it's permanent id as 11th entry in MSP_BOXIDS (27)

Cleanflight internally uses another indexing - `boxId_e`) - which does not contain obsolete permanent ids

Consider looking at https://github.com/cleanflight/cleanflight/blob/master/src/main/fc/msp_server_fc.c#L387. I did reimplement box handling in cleanflight, responses are the same, but it may be easier to understand the mappings ..

christianrauch, 2017-02-04:
@ledvinap Thanks for the explanation.
The enum [boxId_e](https://github.com/cleanflight/cleanflight/blob/master/src/main/fc/rc_controls.h#L52) seems to store all possible modes and their ids. But because `BOXVARIO` is commented, they do not match with the ids in `MSP_BOXIDS`. How are these mode ids related?
What are the permanent id in `MSP_BOXIDS` used for? E.g. for what reason would a client request those ids?

ledvinap, 2017-02-04:
boxId_e is never sent over MSP and is 'unrelated' to permanentId.

`box_t boxes` (https://github.com/cleanflight/cleanflight/blob/master/src/main/fc/msp_server_fc.c#L138 ) is  used to map `boxId_e` to `permanentId`.
MSP_BOXIDS return `permanentId`, it is guarantied to be constant for given box.
You can assume that permanentId 27 is always FAILSAFE, but it may be sent in different positions in `MSP_BOXIDS` reply and thus different bit position in `MSP_STATUS` (and may have different numerical value in `boxId_e`)

hydra, 2017-02-05:
ok, closing this, nothing to do

closed 2017-02-05 by hydra