cleanflight/cleanflight #2611: Autonomous Flight without any RC input

dyereh, 2017-02-08:
I have been reading & working through this project for the past several months & am trying to send commands to the controller by directly accessing rcCommands[] in the fc. I have read thru the other 'autonomous' issues on the project but have some specific questions.

Would there be some easy(ish) way to just get rid of the rx module in cleanflight (or at least have the fc completely ignore rx). We are sending commands thru our own wireless embedded module using msp to directly modify rcCommands (and also use mwArm/mwDisarm).

Currently I have just modified the luxfloat controller by replacing rcCommand[] with my own new array & I update these commands through msp. I have bypassed all arming flags so that the motors are always armed (I realized this is not the safest implementation but is a quick temporary fix). This method seems to work for the most part but I am experiencing hard to repeat glitches (sometimes fc enters cli probably because some bad data is sent to serial interface. This could be resolved by arming properly so cli is ignored). The bigger issue is that once in a while the red status light on the seriouslypro F3 just turns off and won't turn back on without a power cycle.

Sorry for the rant but tl;dr - trying to send commands directly to controller via MSP (NOT sending roll/pitch/yaw cmds as we don't want any filtering or remapping). Any suggestions on the simplest approach to completely bypassing rx without 'breaking' anything. 

If I can get this working robustly I would love to share the mods with the community :)

martinbudden, 2017-02-08:
What are you actually trying to achieve? In particular why is the MSP receiver type not suitable?

Channel remapping is easily turned off, as is rc command smoothing.

dyereh, 2017-02-08:
It may in fact be suitable and I may be over complicating things. I would like to affect the code minimally (or not at all if possible). With the MSP_SET_RAW_RC - is there a way that there is 0 filtering on the RC data as well as no remapping (so if I send roll = 1550 this is linearly mapped directly to rcCommand[])?

martinbudden, 2017-02-09:
If you set `rc_smoothing` off and `tpa_breakpoint` appropriately I think you should get what you want. (I'm not 100%, sure though, you'd need to try it out and/or check the code).

dyereh, 2017-02-09:
Thanks for all the help martin. I can now send the commands through wireless MSP, there are just a couple things I am noticing (and maybe have to trace through the code further to figure out)

1. When sending MSP_SET_RAW_RC (I do so wirelessly over port 3) I notice the red light on the SPRacingF3 turns off instead of doing the usual on/off flashing. When armed it goes solid. Just wondering if this is the expected behavior because I remember it still flashing on and off normally when disamed & receiving RX (ie Spektrum sat).

2. Through much trial and error I found that I need to send around 50-100 MSP messages of RC before putting the aux 1 HIGH (say around 1600) to arm the fc. I found that if I simply started by sending command with throttle low & aux 1 high the motors would twitch and then fail to arm when I sent subsequent commands.

Right now the pseudocode for this would look like:

for i = 1 to 100: 
     sendMSPCommand #200 - > [1500, 1500, 1050, 1500, 1000, 1000, 1000, 1000] //not arming 

sendMSPCommand #200 - > [1500, 1500, 1050, 1500, 1600, 1000, 1000, 1000] //arm fc

while True:
      sendMSPCommand #200 - > [roll, pitch, throttle, yawRate, 1600, 1000, 1000, 1000] //start sending RC commands

I'm wondering if it the reason I have to initially send rc data without arming is due to some failsafe mechanism?

Thanks :)

closed 2017-02-25 by hydra