cleanflight/cleanflight #2612: can't arm motors!

waynewendrone, 2017-02-09:
I have a sp racing f3 configured on clean flight and it only arms the motors when the usb cable is plugged in and clean flight is connected to the board.
I'm using devo 10 and rx1002 pls help, everything else seems fine and I've tried to change the minimum throttle and have reset the board many times

imtjuanjose, 2017-02-09:
Check the System Load with the 'status' command in CLI from configurator, just to check that it's not arming due to system overload.

Calibrate the sensors and make sure that the quad is in a flat surface while arming. (If your board or sensor board is mounted at some angle, make sure to compensate for that in the Board and Sensor Alignment)

If you are not sure if it's a problem with the Throttle Ranges, you can try using a switch to arm the quad.

Greetings 
Juan José

waynewendrone, 2017-02-09:
I can arm the board while it's connected to cleanflight but can't as soon as I disconnect USB cable and re plug li-po

imtjuanjose, 2017-02-09:
I see, there was a similar problem [here](https://github.com/cleanflight/cleanflight/issues/2572).

Maybe you can post the output of DUMP and STATUS.

waynewendrone, 2017-02-09:
If I did would you be able to help?

waynewendrone, 2017-02-09:
Any idea why this is the case?

imtjuanjose, 2017-02-09:
I will probably not be able to help you if it's a bug or a Software Issue because I'm new using cleanflight, but I'm almost sure that anyone that can help you will require that info to help you.

Also have you check that your BEC is powering the Board? I found a post in another page where someone had the same Issue and that was the problem.

waynewendrone, 2017-02-10:
I'm powering the board using a matek pdb

waynewendrone, 2017-02-10:
Through the esc servos

waynewendrone, 2017-02-11:
I'm not getting any beeps from the esc when I power on without cleanflight but if I plug the board into cleanflight first and then plug li-po I will hear beeps from the esc and can proceed to arming the board

waynewendrone, 2017-02-11:
if anyone could help, here arm my current dump data

waynewendrone, 2017-02-11:
dump

# version
# Cleanflight/SPRACINGF3 1.14.2 Dec 31 2016 / 02:11:08 (747971f)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -AMPERAGE_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -OSD
feature VBAT
feature MOTOR_STOP
feature FAILSAFE
feature RX_PARALLEL_PWM

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# led
led 0 15,15:ES:AI:0
led 1 15,8:E:FW:0
led 2 15,7:E:FW:0
led 3 15,0:NE:AI:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:AI:0
led 7 0,7:W:FW:0
led 8 0,8:W:FW:0
led 9 0,15:SW:AI:0
led 10 7,15:S:FW:0
led 11 8,15:S:FW:0
led 12 7,7:U:FW:0
led 13 8,7:U:FW:0
led 14 7,8:D:FW:0
led 15 8,8:D:FW:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
[... 314 lines cut ...]
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

hydra, 2017-02-14:
please use rcgroups for user-support

closed 2017-02-14 by hydra