cleanflight/cleanflight #2616: Documentation - CLI.md is outdated

sansha, 2017-02-11:
while upgrading from 1.13 to 1.14.2 I dumped my settings and tried to carefully restore them manually because I thought just dumping them back in could erase the new defaults which seem to be useful.
I noticed that many values in [CLI.md](https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md) are outdated, e.g. 
- gyro_soft_lpf which isn't present anymore
- I'm not sure how motor_pwm_rate is used because I'm quite sure with gyro_sync and oneshot125 (or even faster) 400 is not correct but above 500 it switches to brushed mode.
- looptime is still present in this doc even though it was apparently removed from 1.14.2

it would also be helpful if there would be some Description on how and why the DEFAULTS changed.
On my Quad I need a high D term value and it would be great if I knew if I should stick with that or with the new defaults due to some implementation changes.

andre92, 2017-02-14:
Default lowpass settings have been raised and require careful tuning of the notch filter or you burn your motors. The new defaults for the notch filter do not produce a working filter (notch_cutoff should be below notch_hz, not the same)
Your safest bet is using the old defaults. Oneshot125 is synced with pid loop and ignores motor_pwm_rate. If your ESCs do not support Oneshot125, just leave it default.

hydra, 2017-02-14:
@andre92 any recommendations on new defaults?

andre92, 2017-02-15:
@hydra it will be difficult to find one default that fits all classes of multicopter. I would use the old defaults, as they are at least safe for all copters, and add some presets either in wiki or in configurator.
iNav has made some presets for different craft, including pid settings
for racers:
asynchronous gyro/acc processing
gyro hardware lpf: 256hz / off
gyro soft lpf 90
dterm soft lpf 100
notch cutoff 100
notch hz 200
default pids from Betaflight 3.1

for large multicopters:
synchronous processing
gyro hardware lpf: 42hz
gyro soft lpf: 60
dterm soft lpf 40hz
notch cutoff 40
notch hz 80
old default pids

it should be noted that racers will fly with pretty much any settings, however large copters will smoke motors with defaults for racers. I suggest using defaults for large copters.

sansha, 2017-02-15:
Thank you for the detailed input.
Can I somehow help in also updating the Documentation?
I could maybe also write a Documentation page about the notch filter as there's little info about it out there if you could provide me with the corresponding pull requests?
I tried to setup the filter but got very confused because the only info about it seems to be here:
https://www.google.de/amp/www.propwashed.com/notch-filter-practical-guide/amp
And the cleanflight GUI differs quite heavily from this description.

Also a good idea to keep the defaults to values which cannot harm

McGiverGim, 2017-02-17:
The configuration options for the notch filter are ready in the development gui?  I didn't see it. I'm very interested to try it.

sansha, 2017-02-17:
They are apparently in the betaflight GUI, not in the cleanflight GUI.

andre92, 2017-02-20:
@McGiverGim currently Cleanflight only has one notch filter and it is only available in CLI.
iNav and BF have them configurable in GUI (except iNav doesn't support notch on F1)

McGiverGim, 2017-02-20:
@andre92 thanks! Very clear all. I will try it.

Thanks again!

closed 2017-02-25 by hydra