cleanflight/cleanflight #2617: I have lidar-lite v3 code. Would anyone like support for this?

marksabido, 2017-02-13:
Hi,

A few months back I wrote some code for the stm32f4 to control the above lidar using the i2c interface. I have just started looking for a new FC board and stumbled across this marvellous project. I was just wondering if there would be any interest in adding support for this lidar?

I've spent a couple of years developing commercial real time code for the stm32fx processors so I am reasonably familiar with them.

tednv, 2017-02-20:
it would certainly beat what everyone is wasting their time with now (barometric and sonar sensors) although the $150 garmin price sticker may be somewhat off-putting to most. but then again, if they do not lower their price, our friend in china may decide to do something about it if you know what I mean

grab an f3 board its cheap and will do i2c, pwm and will have plenty of ports to spare

don't bother with cleanflight or betaflight btw, inav is where it's at. cleanflight or betaflight are more for the acro flyers that like to go a million miles per hour while pretending to have control over their craft and then bragging about how they spent more than 45 seconds in the air without crashing

there are people already using lidar with arduino code, should be able to port it over without much effort

martinbudden, 2017-02-20:
@marksabido , the lidar code is a better match for iNav. A PR there would be welcome. See https://github.com/iNavFlight/inav

marksabido, 2017-02-20:
Thanks for the replies. 

Makes sense what you are saying, I will check the inav project

closed 2017-02-25 by hydra