cleanflight/cleanflight #2622: How to disable bottom right arming?

ben-j69, 2017-02-20:
I just got an e010s and to arm it I have to bring the left stick on bottom right.

I want to use an aux channel to arm it, how to remove the bottom right stick arming?

andre92, 2017-02-21:
it gets automatically disabled when you set up a mode to arm by aux, no need to disable it yourself.

ben-j69, 2017-02-21:
I did try to set up the arm mode but it never go to green despite I am in the good range.

andre92, 2017-02-21:
Are your aux channels working in receiver tab? Maybe you haven't set up switches to aux channels in your transmitter.

ben-j69, 2017-02-23:
My AUX channels 1 and 2 works but despite I am in the good range it does not arm. If I set up arm mod on aux 1 I can't arm it anymore with the stick. 

Dump 

Entering CLI Mode, type 'exit' to return, or 'help'

# version
# Cleanflight/SPRACINGF3 1.13.0 Jun  6 2016 / 00:03:10 (a77bc76)
# dump

# version
# Cleanflight/SPRACINGF3 1.13.0 Jun  6 2016 / 00:03:10 (a77bc76)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature RX_PPM
feature MOTOR_STOP
feature FAILSAFE

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
[... 307 lines cut ...]
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 91
set rc_expo = 74
set rc_yaw_expo = 0
set thr_mid = 65
set thr_expo = 6
set roll_rate = 100
set pitch_rate = 100
set yaw_rate = 60
set tpa_rate = 60
set tpa_breakpoint = 1700

#

closed 2017-02-25 by hydra