cleanflight/cleanflight #2623: I2C_ClockSpeed Questions

elbowdonkey, 2017-02-21:
As far as I can tell I have two options when it comes to i2c clock speed: 400khz or 800khz (the latter being the overclocked speed). I'm not an i2c expert, but have a general sense of how it works.

I have an OLED module I'd like to use that appears to want an i2c clock speed of 100khz. When used at 100khz on another microcontroller the display works fine. With the same init commands, etc, it doesn't work on a naze32 acro rev6. The only difference I can see so far is their clock speeds. The 100khz value seems important for another reason as well - in the sample code for the display there's a comment saying "100kHz normal mode, this value must be used for a genuine WMP" - (I have no idea what a WMP is).

Would it be unwise for me to attempt trying to support this slower clock speed?

Is the clock speed thing a red herring?

My general plan was to leave the `i2c.I2C_ClockSpeed` values as they were while instead modifying `display_ug2864hsweg01.c` to use 100khz instead. Before determining what that would really entail I wanted a bit of a sanity check. Can I tell the i2c bus to use different clock speeds for different addresses on the bus? If I set the non-overclocked speed to 100khz, what would the consequences be?

martinbudden, 2017-02-21:
I2C supports clock stretching to allow slower devices to run on a higher clocked bus. @digitalentity recently rewrote much of the I2C code, I'm not sure if iNav supports clock stretching - @digitalentity  - would you care to comment.

I advice would be to just try it and see before you look at altering the code.

elbowdonkey, 2017-02-21:
By "just try it and see" I assume you mean see if the OLED display works. It doesn't, which is the reason I'm pursuing this. On other micro-controllers it works fine because I can very easily set the I2C clock speed.

I'll try just setting the clock speed to 100khz today and see what happens. I was just worried that by doing that it might get the display working, but I'd find out later on that it messed with other sensors (while at full throttle 30 feet off the ground, for example).

elbowdonkey, 2017-02-21:
I suspect my issue is less about i2c clock speed, and more likely pull up resistors. After setting the clock speed to 100k in both cleanflight and iNav with no results I realized there may be another difference I wasn't considering.

On the other micro controllers I tested the display with I was able to easily toggle internal pull ups. I can get the display to not work on those micro controllers if I don't set those pull ups to high. I don't even know where I'd look to do the same thing for the naze32 in software (or if the mcu it uses even supports internal pullups).

Instead, I'm going to add some 4.7k resistors to the display to see if that helps. I'll test it on an Arduino Uno or something first to ensure that with the resistors added, and internal pull ups off that it works as expected, then switch it over to the naze.

elbowdonkey, 2017-02-21:
Oh, I should mention that in iNav when the default i2c clock speed was set to 100k the responsiveness of the GUI was crazy slow. So at the very least it's clear that setting it so low compared to the default has the consequence of a really sluggish GUI.

elbowdonkey, 2017-02-21:
I tested the display with external resistors on an Uno and verified that it worked properly without setting the Uno's internal pull ups. 

When tested on a naze32 it didn't appear to work. What I did see, however, was a spike in i2c errors. I didn't see those errors before so I suspect that the external pull ups might be working, but there's a software issue now. I'll investigate a little bit but I'm close to actually just giving up. I wanted a tiny display to be able to easily see what flight mode the flight controller is in. I can see that same info using an on the go USB cable and an android device, so will probably fall back to that option.

elbowdonkey, 2017-02-24:
I gave up on this. Closing.

closed 2017-02-24 by elbowdonkey