cleanflight/cleanflight #2626: min_command resetting

elbowdonkey, 2017-02-24:
I'm baffled by an issue I can repeat without fail on a naze32 acro rev 6.

When powered by USB I set the `min_command` setting to 930 (not through the CLI, but the Configuration tab). I can save the setting, reboot, even unplug and replug USB and the setting is 930 upon reconnecting.

I use an AUX switch to arm and its low end (which I use to arm) is 1084. Arming works fine on USB power.

If I remove USB power and then plug in a battery I'm not longer able to arm. Removing battery power and switching to USB I see that the `min_command` value has reset to whatever the default is (which I think is 1100 - all I know is that it's higher than 1084).

 As far as I can tell all other values seem to be as I set them.

What could be going on here?

elbowdonkey, 2017-02-24:
Here's a `dump`:

# dump

# version
# Cleanflight/NAZE 1.14.2 Dec 31 2016 / 02:12:34 (747971f)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -AMPERAGE_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -OSD
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature LED_STRIP
feature ONESHOT125

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 0,0:D:A:0
led 1 1,0:D:F:2
led 2 2,0:D:CW:10
led 3 3,0:D:CW:10
led 4 4,0:D:F:2
led 5 5,0:D:A:0
led 6 0,0::C:0
led 7 0,0::C:0
led 8 0,0::C:0
led 9 0,0::C:0
led 10 0,0::C:0
led 11 0,0::C:0
led 12 0,0::C:0
led 13 0,0::C:0
led 14 0,0::C:0
led 15 0,0::C:0
led 16 0,0::C:0
led 17 0,0::C:0
led 18 0,0::C:0
[... 319 lines cut ...]
set gtune_loP_rll = 10
set gtune_loP_ptch = 10
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

closed 2017-02-25 by hydra

elbowdonkey, 2017-02-27:
I'm curious why this was closed. It's still an issue I can easily recreate on a very common setup.

brycedjohnson, 2017-03-01:
The betaflight code was all merged into Cleanflight recently, it was a huge set of changes, so a lot of issues have been fixed/gone away/changed.  Might need to retest this on the next cleanflight release

hydra, 2017-03-01:
yes, if it's still an issue please create a new issue and refer to this one.

commit 398df910 (2017-03-12): Fix #2626

Fix swapped terms in function documentation. #2626
commit 45408197 (2017-03-12): Merge pull request #2627 from GTorreil/patch-1

Fix #2626