cleanflight/cleanflight #2634: Hexacopter + Oneshot125 on CC3D

olegccc, 2017-03-01:
If I choose these two options together on CC3D the board stops working. It blinks four times on startup and that's it. USB is not recognized. Reflashing helps but once I select these two options together it again gets into the same state. I can choose Oneshot125 or Hexacopter (X, +) but not these options together. Tried different versions starting from 12.

andre92, 2017-03-01:
https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20CC3D.md
It appears the board doesn't fully support hexacopters due to hardware limitations. Something to do with the pin mapping. Did it work as a hexacopter without Oneshot?

olegccc, 2017-03-02:
CC3D works good with hexacopter on Librepilot hardware. It even has support of Oneshot125 there which works too. I checked yesterday with Betaflight and it at least doesn't brick the board when I choose Oneshot125 - but engine doesn't spin if I choose it. And it works with PWM option on Betaflight.

hydra, 2017-03-02:
Tlpoease retest with 2.0.0-RC1

olegccc, 2017-03-03:
@hydra I checked here - https://github.com/cleanflight/cleanflight/releases - it has no CC3D hex file for 2.0.0

olegccc, 2017-03-03:
It seems like comment regarding hexacopter mode in CC3D board description is old. I checked target/CC3D/target.h and it has the following definition:

`#define TARGET_MOTOR_COUNT 6`

However versions 1.14 and 1.13 don't work in hexacopter mode - I cannot arm them. It blinks ~16 times and doesn't switch to armed mode. I checked possible conditions in source code, seems like it doesn't pass overload check. I flashed version 1.12 and it works ok - I can arm it and it turns on the motors.

MJ666, 2017-03-04:
The last looks like an classic timer clash issue. Looks to be the ppmAvoidPWMTimerClash() function is not used anymore for any of the targets? This likely will also affect a few more targets which are using PPM and PWM receivers. Does this also happen when an serial RX protocol is used? Which release of CleanFlight you are using?

On the Alienflight F3 PPM input is actually only working as long only 4 motors are used. For 6 or 8 motors the PPM input is not working. Motor 6 is sharing the timer with PPM in this case.

olegccc, 2017-03-04:
@MJ666 The board is controlled over MSP connected to mini USB. ARM mode is triggered thru AUX1. I was able to launch and land my hexacopter using CC3D v.1.12 so it definitely works with it.

hydra, 2017-03-05:
@MJ666 Can you have look into what became of the 'ppmAvoidPWMTimerClash' code, it's the probable cause.  

@olegccc and @MJ666 Perhaps check with the betaflight guys?

Certainly we won't be looking to fix this for v1.x and we should try and find a fix for 2.x if the issue is still present.

@olegccc can you build a 2.x binary and test it?

olegccc, 2017-03-05:
@hydra 
2017-03-05 @ 8:49:22 PM -- Serial port successfully opened with ID: 266
2017-03-05 @ 8:49:23 PM -- MultiWii API version received - 1.32.0
2017-03-05 @ 8:49:24 PM -- Flight controller info, identifier: CLFL, version: 2.0.0
2017-03-05 @ 8:49:24 PM -- Running firmware released on: Mar 5 2017 19:30:14
2017-03-05 @ 8:49:32 PM -- No configuration received within 10 seconds, communication failed

olegccc, 2017-03-05:
@hydra I tried to use betaflight. When I select ARM mode in configurator it doesn't provide me possibility to choose channel number - channel selector is presented on the screen but it has empty list. I noticed it is only MSP receiver issue - when I choose other receiver I can see AUX1 as possible choice. I tried to choose first AUX1 and then switch to MSP but anyway when I send channels their status is not reflected in the configurator (like it works in cleanflight) and it never switches to armed mode despite I use the same data as for cleanflight.

mikeller, 2017-03-06:
@olegccc: Did you use the Betaflight configurator?

olegccc, 2017-03-06:
@mikeller Yes, sure.
I also tried to use inav but it has other issues - it allows to select hexacopter for cc3d and save it but once it reconnects it shows again default choice - quad x. Tried different versions.
So after trying to make betaflight, cleanflight and inav working with hexacopter on cc3d i found only one combination which works - cleanflight but not latest version. New versions and any version of betaflight/inav don't work, due to different reasons.

olegccc, 2017-03-07:
Regarding last message - I figured out why it doesn't work - simply because MSP is disabled by default for Betaflight/INAV. Now it works.

olegccc, 2017-03-07:
However there remains the same issue with Cleanflight - it works only with old versions.

mikeller, 2017-03-08:
cleanflight V2 is based off Betaflight, so if it doesn't work in Betaflight it won't in cleanflight V2. Also, you didn't make it clear to start with that MSP RX was part of your setup, that was a bit confusing. And yes, this, and other less common RX protocols have been disabled in Betaflight for F1 based boards, to make the firmware fit into the available flash space.

olegccc, 2017-03-08:
@mikeller the issue discussed at the beginning was without setting MSP. You only need to set both hexacopter+Oneshot125. I can create separate issue which is for MSP+hexacopter on CC3D never arming.

hydra, 2017-03-16:
ok, closing then, unsupported configuration and we won't be doing another v1.x release due to lack of developer time while we focus on CF v2.x.

closed 2017-03-16 by hydra