cleanflight/cleanflight #2640: make test doesnt work

error414, 2017-03-05:
When I tried run tests I got this error

`gcc version 5.4.0 (GCC)`

with 6.3 version I got same error

```
../../obj/test/maths_unittest
Running main() from gtest_main.cc
[==========] Running 7 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 7 tests from MathsUnittest
[ RUN      ] MathsUnittest.TestConstrain
[       OK ] MathsUnittest.TestConstrain (0 ms)
[ RUN      ] MathsUnittest.TestConstrainNegatives
[       OK ] MathsUnittest.TestConstrainNegatives (0 ms)
[ RUN      ] MathsUnittest.TestConstrainf
[       OK ] MathsUnittest.TestConstrainf (0 ms)
[ RUN      ] MathsUnittest.TestDegreesToRadians
[       OK ] MathsUnittest.TestDegreesToRadians (0 ms)
[ RUN      ] MathsUnittest.TestApplyDeadband
[       OK ] MathsUnittest.TestApplyDeadband (0 ms)
[ RUN      ] MathsUnittest.TestFastTrigonometryATan2
atan2_approx maximum absolute error = 7.152557e-07 rads (4.098114e-05 degree)
[       OK ] MathsUnittest.TestFastTrigonometryATan2 (0 ms)
[ RUN      ] MathsUnittest.TestFastTrigonometryACos
acos_approx maximum absolute error = 6.760856e-05 rads (3.873685e-03 degree)
[       OK ] MathsUnittest.TestFastTrigonometryACos (1 ms)
[----------] 7 tests from MathsUnittest (1 ms total)

[----------] Global test environment tear-down
[==========] 7 tests from 1 test case ran. (1 ms total)
[  PASSED  ] 7 tests.
g++ -g -Wall -pthread -Wextra -ggdb3 -O0 -DUNIT_TEST -isystem ../../lib/test/gtest/inc -MMD -MP -std=gnu++11 --coverage -Iunit -I../main -c uni                                                                                                                       t/parameter_groups_unittest.cc -o ../../obj/test/parameter_groups_unittest.o
In file included from unit/parameter_groups_unittest.cc:27:0:
../main/config/parameter_group.h:258:5: sorry, unimplemented: non-trivial designated initializers not supported
     }                                                                   \
     ^
unit/parameter_groups_unittest.cc:36:1: note: in expansion of macro ‘PG_RESET_TEMPLATE’
 PG_RESET_TEMPLATE(motorConfig_t, motorConfig,
 ^
../main/config/parameter_group.h:258:5: sorry, unimplemented: non-trivial designated initializers not supported
     }                                                                   \
     ^
unit/parameter_groups_unittest.cc:36:1: note: in expansion of macro ‘PG_RESET_TEMPLATE’
 PG_RESET_TEMPLATE(motorConfig_t, motorConfig,
 ^
../main/config/parameter_group.h:258:5: sorry, unimplemented: non-trivial designated initializers not supported
     }                                                                   \
     ^
unit/parameter_groups_unittest.cc:36:1: note: in expansion of macro ‘PG_RESET_TEMPLATE’
 PG_RESET_TEMPLATE(motorConfig_t, motorConfig,
 ^
../main/config/parameter_group.h:258:5: sorry, unimplemented: non-trivial designated initializers not supported
     }                                                                   \
     ^
unit/parameter_groups_unittest.cc:36:1: note: in expansion of macro ‘PG_RESET_TEMPLATE’
 PG_RESET_TEMPLATE(motorConfig_t, motorConfig,
 ^
../main/config/parameter_group.h:258:5: warning: missing initializer for member ‘motorConfig_s::mincommand’ [-Wmissing-field-initializers]
     }                                                                   \
     ^
unit/parameter_groups_unittest.cc:36:1: note: in expansion of macro ‘PG_RESET_TEMPLATE’
 PG_RESET_TEMPLATE(motorConfig_t, motorConfig,
 ^
make[1]: *** [Makefile:809: ../../obj/test/parameter_groups_unittest.o] Error 1
make[1]: Leaving directory '/cygdrive/c/Users/Uzivatel/Documents/copter/aRCiTimerDeploy/cleanflight/src/test'
make: *** [Makefile:1209: test] Error 2

error414@nb /cygdrive/c/Users/Uzivatel/Documents/copter/aRCiTimerDeploy/cleanflight
```

AndersHoglund, 2017-03-07:
You must use clang. GCC can not handle some of the constructs in the code, clang can.

    export CC=clang
    export CXX=clang++
    make test

ledvinap, 2017-03-07:
In this case defining initializer in identical order as struct declaration may fix the error:
```
PG_RESET_TEMPLATE(motorConfig_t, motorConfig,
    .dev = {.motorPwmRate = 400}
    .minthrottle = 1150,
    .maxthrottle = 1850,
    .mincommand = 1000,
);

error414, 2017-03-07:
If I use clang, I get next errors

```
clang++ -g -Wall -pthread -Wextra -ggdb3 -O0 -DUNIT_TEST -isystem ../../lib/test                                                                                                                            /gtest/inc -MMD -MP -std=gnu++11 --coverage -Iunit -I../main -c unit/parameter_g                                                                                                                            roups_unittest.cc -o ../../obj/test/parameter_groups_unittest.o
clang -g -Wall -pthread -Wextra -ggdb3 -O0 -DUNIT_TEST -isystem ../../lib/test/g                                                                                                                            test/inc -MMD -MP -std=gnu99 --coverage -Iunit -I../main -c ../main/config/param                                                                                                                            eter_group.c -o ../../obj/test/config/parameter_group.o
clang++ -g -Wall -pthread -Wextra -ggdb3 -O0 -DUNIT_TEST -isystem ../../lib/test                                                                                                                            /gtest/inc -MMD -MP -std=gnu++11 --coverage -Wl,-T,unit/parameter_group.ld -Wl,-                                                                                                                            Map,../../obj/test/../../obj/test/parameter_groups_unittest.map ../../obj/test/p                                                                                                                            arameter_groups_unittest.o ../../obj/test/config/parameter_group.o ../../obj/tes                                                                                                                            t/gtest_main.a -o ../../obj/test/../../obj/test/parameter_groups_unittest
../../obj/test/gtest_main.a(gtest-all.o): In function `std::operator|(std::_Ios_                                                                                                                            Openmode, std::_Ios_Openmode)':
/usr/lib/gcc/x86_64-pc-cygwin/5.4.0/include/c++/bits/ios_base.h:130: multiple de                                                                                                                            finition of `std::operator|(std::_Ios_Openmode, std::_Ios_Openmode)'
../../obj/test/parameter_groups_unittest.o:/usr/lib/gcc/x86_64-pc-cygwin/5.4.0/i                                                                                                                            nclude/c++/bits/ios_base.h:130: first defined here
../../obj/test/gtest_main.a(gtest-all.o): In function `GetDefaultFilter':
/cygdrive/c/Users/Uzivatel/Documents/copter/aRCiTimerDeploy/cleanflight/src/test                                                                                                                            /../../lib/test/gtest/src/gtest-all.cc:1669: multiple definition of `testing::As                                                                                                                            sertionResult::operator bool() const'
../../obj/test/parameter_groups_unittest.o:/usr/lib/gcc/x86_64-pc-cygwin/5.4.0/i                                                                                                                            nclude/c++/iostream:74: first defined here
../../obj/test/gtest_main.a(gtest-all.o): In function `GetDefaultFilter':
/cygdrive/c/Users/Uzivatel/Documents/copter/aRCiTimerDeploy/cleanflight/src/test                                                                                                                            /../../lib/test/gtest/src/gtest-all.cc:1669: multiple definition of `testing::As                                                                                                                            sertionResult::message() const'
../../obj/test/parameter_groups_unittest.o:/usr/lib/gcc/x86_64-pc-cygwin/5.4.0/i                                                                                                                            nclude/c++/iostream:74: first defined here
../../obj/test/gtest_main.a(gtest-all.o): In function `GetDefaultFilter':
/cygdrive/c/Users/Uzivatel/Documents/copter/aRCiTimerDeploy/cleanflight/src/test                                                                                                                            /../../lib/test/gtest/src/gtest-all.cc:1669: multiple definition of `testing::As                                                                                                                            sertionResult::failure_message() const'
../../obj/test/parameter_groups_unittest.o:/usr/lib/gcc/x86_64-pc-cygwin/5.4.0/i                                                                                                                            nclude/c++/iostream:74: first defined here
../../obj/test/gtest_main.a(gtest-all.o): In function `GetDefaultFilter':
/cygdrive/c/Users/Uzivatel/Documents/copter/aRCiTimerDeploy/cleanflight/src/test                                                                                                                            /../../lib/test/gtest/src/gtest-all.cc:1669: multiple definition of `testing::Te                                                                                                                            st::Setup()'
[... cut ...]                                                                ternal::scoped_ptr<std::basic_stringstream<char, std::char_traits<char>, std::al                                                                                                                            locator<char> > >::reset(std::basic_stringstream<char, std::char_traits<char>, s                                                                                                                            td::allocator<char> >*)'

```

AndersHoglund, 2017-03-08:
Did you clean up after gcc before using clang? `rm -rf obj` is rather effective.

AndersHoglund, 2017-03-08:
@ledvinap only reordering does not help. Tried that. gcc requires all struct elements, in correct order. Including all elements if the ` motorDevConfig_t dev` structure. 
There are also some new ibus tests in the betaflight repo using these sparse initializers that requires clang.

hydra, 2017-03-09:
please create a PR to fix up any tests @martinbudden may also be interested in this.

and yes, we had the same problem before in CF 1.x and the solution then, IIRC, was to specify all values,  you can double check the master-v1.x branch.

AndersHoglund, 2017-03-09:
Betaflight is now using clang explicitly, https://github.com/betaflight/betaflight/pull/2589
As Travis seems to use clang by default, implicitly, it makes sense to use the same locally. If going for gcc, then it must be made explicitly so even Travis uses gcc. Either gcc or clang, not leaving it to some default environment preference. All in the name of consistency.

hydra, 2017-03-16:
CF uses clang for unittests now too.

closed 2017-03-16 by hydra