cleanflight/cleanflight #2654: Missing documentation to generate a "personalized" version

McGiverGim, 2017-03-17:
Hi! I don't know if this is an "issue". If not, please don't hesitate to close this.

I think that now that Cleanflight is based in betaflight, some FC have loose functionalities, like GPS or iBus Telemetry on NAZE for example.

Some of ours, can build the HEX from the code withouth problems, and I know that modifying the "defines" in the target/common.h file can add or disable features. The problem is that we're not sure of dependencies and sizes of each feature.

For example:
- Can I disable (I'm using FlySky Serial RX iBus and softserial in NAZE32):
USE_PPM
USE_PWM
USE_SERIALRX_CRSF       // Team Black Sheep Crossfire protocol
USE_SERIALRX_SBUS       // Frsky and Futaba receivers
USE_SERIALRX_SPEKTRUM   // SRXL, DSM2 and DSMX protocol
USE_SERIALRX_SUMD       // Graupner Hott protocol
USE_SERIALRX_SUMH       // Graupner legacy protocol
USE_SERIALRX_XBUS       // JR
LED_STRIP
TELEMETRY_FRSKY
TELEMETRY_HOTT
TELEMETRY_LTM
TELEMETRY_SMARTPORT
USE_SERVOS

- And add:
GPS
TELEMETRY_IBUS

I don't know if some of the removes can affect other features and how much space (approximatelly) is saved in each remove. A little document with this information will be great, and a lot of users will be happy again ;)

My english is not very good, but if you have this info somewhere I can make the first version of the document.

Thanks for all!

mikeller, 2017-03-17:
It's a bit difficult to come up with reliable numbers for all of these features, since the space optimisation used in the build will allow for reduction of size based on code that is shared between individual features.

If you are willing to go to the length of setting up your own build environment, probably the easiest way to go on about this is to just give it a go, and see if the firmware with all the features you want can be built. I think if you remove all of the RX  / telemetry protocols that you know you don't need (in reality you'll only ever be using one of each at a time), this should go a long way to make space for other things.

Re building, have a look at https://github.com/betaflight/docker-betaflight-build  , this could help you with building your own targets.

McGiverGim, 2017-03-17:
Ok, I will try and give some numbers, to have some data if someone has the same problem as me.

If this works, maybe I can make a doc explaining this possibility, to advanced users.

Regards!

McGiverGim, 2017-03-17:
I've made a test, disabling and enabling the features that I need.

For the NAZE32:
- The original version (with the latest code of the repo) has a size of 351Kb.
- The original version, adding GPS and IBus telemetry, gives an error (size overflow).
- The "personalized" version with the changes of the first message, is only 315Kb. 

So it seems it worked. I have not tried it. I use in my quadcopter the 1.4.2 version adding this PR:
https://github.com/cleanflight/cleanflight/pull/2554
that was never included into cleanflight, so I've a custom made connection with the RX and it will not work in this 2.x version.
I will ask @unitware to see if he can make a PR for version 2.0 or not.

Regards!

McGiverGim, 2017-03-17:
I've seen the PR was included in BetaFlight in https://github.com/betaflight/betaflight/pull/2639 and was merged into CleanFlight too, so the next week I will try to test it :)

HaukeRa, 2017-03-17:
Just a hint: you are probably measuring the file size of the hex file.

While this is a measure of how large the binary is, it does not tell the exact and real size of the flash contents. Look at the compiler output, it will state the flash size plus useful information like static ram usage, which could be included in your documentation as well.

McGiverGim, 2017-03-17:
I don't know if you are talking about this info:

Original version:
   text    data     bss     dec     hex filename
 126312    1444   18260  146016   23a60 ./obj/main/cleanflight_NAZE.elf

My personalized version:
   text    data     bss     dec     hex filename
 113268    1424   15176  129868   1fb4c ./obj/main/cleanflight_NAZE.elf

I've seen some warnings, maybe I've removed too much. I will take a look to it this week, before trying.

HaukeRa, 2017-03-17:
Exactly that.

Text + Data is flash size, data + bss is (static) ram usage.

McGiverGim, 2017-03-17:
@HaukeRa Thanks! So it seems that the reduction in size and memory is significative.

The only thing that remains is to see if it works as expected 😁

McGiverGim, 2017-03-18:
I've found the origin of the warnings... if I comment the LED_STRIP define, I get some warnings (14, one for each element in the array of target.c of NAZE):

``` 
./src/main/target/NAZE/target.c:28:169: warning: excess elements in struct initializer
     DEF_TIM(TIM2, CH1, PA0,  TIM_USE_PPM | TIM_USE_PWM, 0),
                                                                                                                                                                         ^
./src/main/target/NAZE/target.c:28:169: note: (near initialization for 'timerHardware[0]')
./src/main/target/NAZE/target.c:28:242: warning: excess elements in struct initializer
     DEF_TIM(TIM2, CH1, PA0,  TIM_USE_PPM | TIM_USE_PWM, 0),
                                                                                                                                                                                                                                                  ^
./src/main/target/NAZE/target.c:28:242: note: (near initialization for 'timerHardware[0]')
./src/main/target/NAZE/target.c:29:155: warning: excess elements in struct initializer
     DEF_TIM(TIM2, CH2, PA1,  TIM_USE_PWM,               0),
                                                                                                                                                           ^
./src/main/target/NAZE/target.c:29:155: note: (near initialization for 'timerHardware[1]')
./src/main/target/NAZE/target.c:29:228: warning: excess elements in struct initializer
     DEF_TIM(TIM2, CH2, PA1,  TIM_USE_PWM,               0),
                                                                                                                                                                                                                                    ^
./src/main/target/NAZE/target.c:29:228: note: (near initialization for 'timerHardware[1]')
...
```
If I let the LED_STRIP then all compiles without warnings, giving this size:
```
    text    data     bss     dec     hex filename
  120772    1480   16684  138936   21eb8 ./obj/main/cleanflight_NAZE.elf
```
Slighty bigger than before, but it seems ok for me.
I'm a Java developer, my latest program in C was 12 years ago, so I don't know if the warnings are important or how to eliminate them. Maybe someone can look into it. I will use the version with LED_STRIP in the meanwhile.

hydra, 2017-03-19:
if there's compilation issues when you disable a certain feature then please isolate which feature causes the problem and let us know here.  excess array elements refer to the amount of items in the timer list, see the define `USABLE_TIMER_CHANNEL_COUNT` which may be incorrect for your changes.

McGiverGim, 2017-03-19:
@hydra the feature that gives the problem when I disable it is LED_STRIP. If you disable it you will see the messages.
It seems too that is the feature that more space and memory is consuming, so it will be great if we can disable it.
Regards!

McGiverGim, 2017-03-19:
One thing more: the warning does not seem to come from the number of elements (`USABLE_TIMER_CHANNEL_COUNT`) of the array, it seems to come from the number of elements of the structure of each element. 
Each element has a value like this:
`DEF_TIM(TIM2, CH1, PA0,  TIM_USE_PPM | TIM_USE_PWM, 0),`
This `DEF_TIM` is replaced when compiling, and by the error it seems that the latest parameter is being replaced for spaces os something similar, when `LED_STRIP` is undefined.
Regards!

McGiverGim, 2017-03-19:
Today I've crashed my quadcopter, so will not be able to test this until the repair parts arrive, two or three weeks ☹️
I will test it asap.

Regards!

McGiverGim, 2017-03-20:
I cannot fly, but the electronic is working and I'm playing with the version 2, with and without my changes, using the develpment version of cleanflight-configurator.
I've some problems, I don't know if they're actual limitations or something is wrong in my setup.
The both HEX flash correctly (original and personalized version), and both have the same problem. I'm on a Naze32 board:
- When I enable softserial, only one of them appear in the ports section. Naze32 has two softserial.
- Magetometer is not detected. I enable it in configuration but does not work.
- Flash memory for blackbox not detected. 
- Something is really wrong with ibus receiver. Maybe related to https://github.com/betaflight/betaflight/pull/2639 ? I will do more tests.

Can you say me something about this problems? Is a fault of the configurator and I can try with CLI? Or is fault of the firmware?

Regards!

unitware, 2017-03-20:
I have been flying this on a spracingF3neo, and some short hovers using a NAZE32. Maybe you can try it from there?

(Its based on BF 3.1.6 so you might consider using the betafligth configurator, don knkow if there is any difference though).

https://github.com/unitware/cleanflight/tree/bf-combined-ibus-on-3.1.6

McGiverGim, 2017-03-20:
Hi @unitware, thanks for your help. If you want we can talk about your mod in another thread, to not add a lot of messages here about it ;)

I'm trying to compile your branch, but it gives some compiler errors. Do you want to continue here or we use another thread?

unitware, 2017-03-20:
That built fine for me: https://travis-ci.org/unitware/cleanflight/builds/210546524
Probably you are using a to new arm_sdk (it has been updated since then I think) 

Anyway, I cant help more on that, it's just a baseline to show what I have been flying.

McGiverGim, 2017-03-20:
@unitware Yes, I'm using the newer GCC (the latest required by cleanflight). I paste here the error, maybe this new version is showing something that can be wrong:
```./src/main/telemetry/telemetry.c: In function 'telemetryCheckRxPortShared':
./src/main/telemetry/telemetry.c:115:12: warning: implicit declaration of function 'rxConfig' [-Wimplicit-function-declaration]
         && rxConfig()->serialrx_provider == SERIALRX_IBUS) {
            ^~~~~~~~
./src/main/telemetry/telemetry.c:115:22: error: invalid type argument of '->' (have 'int')
         && rxConfig()->serialrx_provider == SERIALRX_IBUS) {```
Thanks for all!

unitware, 2017-03-20:
ohh, I had forgotten to push the latest there - and the build reported green even if there was a failure.. I have now pushed a new update lets see how she builds: https://travis-ci.org/unitware/cleanflight/builds/212930294

McGiverGim, 2017-03-20:
I've tested your build, some things get better, other not:
- Only one softserial in both builds. So maybe this is a "problem" (or feature) from betaflight to cleanflight.
- Magnetometer not working, again a "problem" (or feature) from betaflight to cleanflight.
- Memory DataFlash for BlackBox working with your version, so this is only a bug in Cleanflight, in Betaflight it works.
- With your version, both, iBus Serial and Telemetry seems to work perfectly. No problem. But in CleanFlight, the ibus telemetry is working, but the receiver not at all. I can move my sticks, nothing moves in the configurator. Maybe the CLI command to activate it is different?

Maybe is good to open different issues, one for each of this problems?

unitware, 2017-03-20:
something like this? https://github.com/betaflight/betaflight/issues/2637

McGiverGim, 2017-03-20:
Yes, something like this. But for cleanflight. I suppose that that your branch is not using latest version of BetaFlight?

unitware, 2017-03-20:
No that branch is a backport upon something that worked at that time (based on BF3.1.6). The issue https://github.com/betaflight/betaflight/issues/2637 is reported on bf-master at that time (I haven't tested it during the weekend, it might be resolved now). You should do a report on cleanflight as well since it has been merged there.

McGiverGim, 2017-03-20:
I've seen the magnetometer is commented in the target.h of NAZE32. It seems to have been disabled to save space. So I will add some info about this in the documentation for a personalized version too. 

I've seen that the SoftSerial maybe has been removed by the same: space. But It's not clear (I didn't see where is defined this in the code, so I've opened an issue too, because is strange that it seems to have only one softserial).

McGiverGim, 2017-03-22:
I close this issue. The PR has been created in https://github.com/cleanflight/cleanflight/pull/2673

I will open an Issue for every other problem found in this thread.

Thanks to all for your great help!

closed 2017-03-22 by McGiverGim