cleanflight/cleanflight #2656: cannot get soft serial to work on sp f3 mini

ZacharyLightman, 2017-03-18:
i am using an spf3 mini and I cannot get the soft serial port 1 (the only one) to function. I have a gps running at 9600 baud (checked with serial port on pc, works brilliantly). I have set this up on a naze board will little difficulty but I cannot make this work in the new mini flight controller to save my life. Is this a known issue?

McGiverGim, 2017-03-23:
What version are you using? The new CleanFlight version 2.x? By your comment, it seems that you're talking about version 1.x. If this is the case, the developers will not support this version, and you will not receive help.

ZacharyLightman, 2017-03-23:
Thank you for the suggestion. I didn't even realize the configurator had to be manually updated. I manually installed the new chrome extension and will try that out.

ZacharyLightman, 2017-03-23:
I am using the new configurator version now. I got my gps and soft serial ports to function! Although I noticed the magnetometer isn't supported?

McGiverGim, 2017-03-23:
Please close the issue, and take a look to: https://github.com/cleanflight/cleanflight/issues/2672

closed 2017-03-23 by ZacharyLightman