cleanflight/cleanflight #2663: Motors out of sync

ThomasH3, 2017-03-19:
Hello. 
I've tried to setup cleanflight on 2x SP Racing F3 and a Naze32. With all controllers I'm facing the same issue ..... If I increase throttle the motors are not accelerating at the same speed.
Usually 1 is increasing to 100 %, 1 is not increasing at all and stays at min throttle, and 2 running at "normal" speed. The problem remains on each controller on the same motors, even if I swap the esc or motor connection on the controller.
Do you have an idea if this issues are software or hardware related

andre92, 2017-03-19:
Maybe it is related to having no propellers on and being bench tested? When you arm any flight controller, it will try to stabilize itself. It is nothing to worry about.

ThomasH3, 2017-03-20:
Thanks, I will test the naze setup on the weekend

SteveAmor, 2017-03-21:
As @andre92 said.  Nothing to worry about. Put the props on and fly it.  Please close this issue.  Issues are not for support.  They clutter the repository and divert developers time.  Many thanks.

closed 2017-03-22 by mikeller