cleanflight/cleanflight #2665: seriusly pro evo upload issue

MauroMombelli, 2017-03-19:
Hello,
I-m working on uploading the firware using an ESP8266 over the USART1.

All sort of work, I can still get bidirectional communication over the USART1, but then i bridge BOOT and reboot the board, the USART1 bootloader remain silent. It has worked for a while, and now i tried wih different PC, windows (demonstrator GUI from ST) and linux (stm32flash) and even using directly a terminal to issue the "0x7F" that should trigger an "y" of 0x1f(? if i remember correctly), but nothing happen.

DFU works fine, uploading the firmware and using cleanflight configurator as no issue.

Any suggestion on what could have happen and how to solve?

mikeller, 2017-03-20:
Have you tried using USART1 after flashing? If so, does it work for non-flashing purposes? If no, it could be a hardware defect.

MauroMombelli, 2017-03-20:
yes,  the USART1 work perfectly running user program uploaded by USB DFU

MauroMombelli, 2017-03-21:
Good news everyone, yesterday i manager to fix tue upload using a bluepill ad target, after thatbibtested again the serously pro evo and worked first time.
Ad the other day i did alla the test in succession, the only problem I can think of is late time (aka human error) or some capacitor.
if you are interested in the procedure: https://sourceforge.net/p/stm32flash/tickets/94/

closed 2017-03-21 by MauroMombelli