cleanflight/cleanflight #2666: NAZE32 DataFlash not detected

McGiverGim, 2017-03-20:
The DataFlash of the NAZE32 appears as missing with the latest code of cleanflight and cleanflight-configurator.

I don't know if it's a problem of the firmware or the configurator. 

To reproduce the problem is easy: enter the configurator with a NAZE32 board. You will see in the top right the message that the DataFlash is not available.

[image]

This problem was detected here: https://github.com/cleanflight/cleanflight/issues/2654 but I think is better to create an issue for it.

McGiverGim, 2017-03-21:
I forgot to mention, that with the latest version of BetaFlight and with CleanFlight 1.4.2 the DataFlash is found and works correctly.

McGiverGim, 2017-03-23:
It seems that the problem are the default values for the NAZE32. In the BlackBox tab, there's a combo for selecting serial or on-board dataflash. By default it was in serial. If I change it, the dataflash is found.

It seems something similar to this: https://github.com/cleanflight/cleanflight/issues/2675

In Betaflight both, Beeper and DataFlash work out of the box.

So I don't know if this is a bug or not. Do I close this issue?

Thanks!

McGiverGim, 2017-03-24:
I've found "the problem" but I need the advice of developers to see how to solve it in a better way.

I've found that the `target.h` of NAZE, is missing this define:

`#define ENABLE_BLACKBOX_LOGGING_ON_SPIFLASH_BY_DEFAULT`

If I add this define to the file, then the BlackBox dataflash is active by default.

The problem is that I think that only the REV6 has a dataflash chip. So maybe it will be better to add this only for REV6. I don't know how to make this.

Some developer can help with this? Or this can be added and if not dataflash is found then it fallbacks to serial? Thanks!

mikeller, 2017-03-24:
It will have to be done in `targetConfiguration`, after the hardware detection has been run. I can look into it.

hydra, 2017-04-01:
@mikeller any update on this?

hydra, 2017-04-01:
to recap:

you need to detect the hardware revision on the naze, then set blackbox to use the flash chip on the naze if it has one, if it's not got one, it should be set to serial by default.

hydra, 2017-04-17:
lets close this, no-one currently has time and the NAZE32 is old and legacy now anyway so it's doubtful anyone will ever work on it.  a PR to update the documentation would be welcomed however.

closed 2017-04-17 by hydra