cleanflight/cleanflight #2668: Only one SoftSerial in NAZE32

McGiverGim, 2017-03-20:
As declared in the documentation: https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20Naze32.md

The NAZE32 can have two SoftSerial when enabled. But if I enable them in the actual master code of CleanFlight, only one of them appears in the ports page.

I've tried BetaFlight too with the same result. There's a closed issue in BetaFlight talking about removing the SoftSerial from some F1 targets (like NAZE32): https://github.com/betaflight/betaflight/issues/2004 but I don't know if this is only for Rev5 NAZE or what's the point of view of developers about SoftSerial in CleanFlight. If they are not supported It's strange that one SoftSerial appears, but the other not.

McGiverGim, 2017-03-20:
I've tested the SoftSerial with a GPS and it works, so it's true: there're only one SoftSerial and it's working.
So it seems that this feature (SoftSerial support) has not been removed, at least not totally.

McGiverGim, 2017-03-21:
After playing with the code, I've found where the SoftSerial 2 is being removed. In the `target/NAZE/initialisation.c`:
```    
   if (hardwareRevision != NAZE32_SP) {
        i2cInit(I2C_DEVICE);
        serialRemovePort(SERIAL_PORT_SOFTSERIAL2);
    } else {
```
If I comment the line with the serialRemovePort, the SoftSerial 2 appears in the configurator again. I don't know if this is a conscient decision and why it has been made. Some developer can clarify this?

Thanks!

scrawny1302, 2017-03-28:
Same for me. I need softserial1 for rgb led and softserial2 for telemetry. Is it possible to change or configure this?

McGiverGim, 2017-03-28:
@scrawny1302 Are you talking about NAZE32? I think that if you activate softserial, then you cannot use the LED_STRIP. I don't know about other targets.

scrawny1302, 2017-03-28:
Yes NAZE32. I have activated softserial but only softserial1 was in the ports tab. I need softserial2 (pins 7 and 8) to send telemetry to my taranis. At the moment I have flashed betaflight. (Sorry for the wrong issuetracker but it seems the same problem because of the same softwarebase)

McGiverGim, 2017-03-28:
You will have the same problem with CleanFlight or BetaFlight, the code that I pasted before is the same for both, and they remove the second softserial. You will need to go to an older version.

And as far as I know, with two softserial, you can't use one for led strip. It's a hardware incompatibility. You must deactivate the softserial to use the led strip feature. If I'm wrong and you make it work I will be very interested because I tried it before without luck.

I can compile for you a version with modified code to have two softserial, but I don't know if this will have other secondary effects.

scrawny1302, 2017-03-28:
Thats the problem, they had a reason to deactivate the second softserial port (flash space). I think I will try it with an older betaflight version to test it first.

McGiverGim, 2017-03-28:
If it's only for flash space, you can deactivate protocols that you don't use and activate the second softserial if you know how to compile the source code.
I've made a document about how to do this: https://github.com/McGiverGim/cleanflight/blob/document-customized-version/docs/Customized%20Version.md

hydra, 2017-04-01:
@McGiverGim please look though the history of the code using git and fit the commit that added serialRemovePort() - then you will be able to see why (commit message) or ask the author.

hydra, 2017-04-01:
ok, i did a bit of digging for you

the original code was here, from me, a long time ago.

https://github.com/cleanflight/cleanflight/blame/3a634ae5c559aebb00d5913f9c1a2b3abd60c387/src/main/main.c

it was recently changed here:

https://github.com/cleanflight/cleanflight/commit/f7518203aac9762cd24736fe7ddaed65d91aeb19

the original code only removed SOFTSERIAL2 if the board was a NAZE32_SP.

IMHO the new code in that commit is in error.

@blckmn would like a comment / review on this please.

hydra, 2017-04-01:
i think the code in initialisation.c should be:

```
    if (hardwareRevision == NAZE32_SP) {
        serialRemovePort(SERIAL_PORT_SOFTSERIAL2);
        
        if (!doesConfigurationUsePort(SERIAL_PORT_USART3)) {
            serialRemovePort(SERIAL_PORT_USART3);
            i2cInit(I2C_DEVICE);
        }
    } else {
        i2cInit(I2C_DEVICE);
    }
```

hydra, 2017-04-01:
PR created.

hydra, 2017-04-06:
PR merged, closing.

closed 2017-04-06 by hydra