cleanflight/cleanflight #2670: Silabs drivers

Carlos1301, 2017-03-21:
Hello,
No matter what I do, I don't seem to be able to run Cleanflight on my laptop due to the drivers. I installed them but as soon as I try to connect I get a message saying that Windows does not recognize the driver.
I have a Lenovo E460 running Windows 7 professional. I am trying to connect a Vortex 150. It has an F3 FC.
Any help would be really appreciated.
Thanks.

lichtl, 2017-03-21:
Hmm I think for the vortex you will need the STM32 VCP driver

Carlos1301, 2017-03-21:
I have tried that one too and nothing....

SteveAmor, 2017-03-21:
@Carlos1301 Please close this issue.  The Github issue track is not for end-user support.

Carlos1301, 2017-03-21:
Ok. I will close it. Could you tell me then where I can get some help?
Thanks.

closed 2017-03-21 by Carlos1301