cleanflight/cleanflight #2672: no MAG on naze32 rev6 10DOF

nightflasher, 2017-03-22:
As the subject already says, the current version of cleanflight is missing the MAG which was detected by older versions of cleanflight.
I don't know if i want to use it but in the near future, but i had used it before and i don't know if anybody else use it. I just wanted to tell.

McGiverGim, 2017-03-22:
BetaFlight deactivated it by size reasons. It is commented in the code. If you know how to compile the code you can activate it again.
Some info here: https://github.com/cleanflight/cleanflight/issues/2654
I don't know if CleanFlight will let it deactivated or not.

McGiverGim, 2017-03-22:
More info: it is commented in `target/NAZE/target.h`, now in the line 104: https://github.com/cleanflight/cleanflight/blob/master/src/main/target/NAZE/target.h#L104 :
```
/*
#define MAG
#define USE_MAG_HMC5883
#define MAG_HMC5883_ALIGN       CW180_DEG
*/
```
If you uncoment this code, the MAG will work again.

hydra, 2017-03-29:
there are new defaults, some documentation has been created to show how to create customized firmware should it be required.  consider upgrading to an F3 board.

closed 2017-03-29 by hydra

nightflasher, 2017-04-03:
ho do i get the 1.14.2 code? now i only get 2.0.0

nightflasher, 2017-04-04:
i have two Afroflight32 boards and tried 1.14.2 on both.
on the first board: no MAG
on the second board: MAG
both have the same hardware:
- MPU6500
- BMA280
- HMC5883

the first board only has MAG on 1.13.0...not sure what's wrong, even the same firmwarefile on both boards game me different behavior.