cleanflight/cleanflight #2674: Can't disable LED_STRIP feature in code, compiler issues

McGiverGim, 2017-03-22:
When I disable the LED_STRIP feature to save some space in the `target/common.h` file, the compiler throws a lot of warnings like this:
```
./src/main/target/NAZE/target.c:28:169: warning: excess elements in struct initializer
     DEF_TIM(TIM2, CH1, PA0,  TIM_USE_PPM | TIM_USE_PWM, 0),
                                                                                                                                                                         ^
./src/main/target/NAZE/target.c:28:169: note: (near initialization for 'timerHardware[0]')
./src/main/target/NAZE/target.c:28:242: warning: excess elements in struct initializer
     DEF_TIM(TIM2, CH1, PA0,  TIM_USE_PPM | TIM_USE_PWM, 0),
                                                                                                                                                                                                                                                  ^
./src/main/target/NAZE/target.c:28:242: note: (near initialization for 'timerHardware[0]')
./src/main/target/NAZE/target.c:29:155: warning: excess elements in struct initializer
     DEF_TIM(TIM2, CH2, PA1,  TIM_USE_PWM,               0),
                                                                                                                                                           ^
./src/main/target/NAZE/target.c:29:155: note: (near initialization for 'timerHardware[1]')
./src/main/target/NAZE/target.c:29:228: warning: excess elements in struct initializer
     DEF_TIM(TIM2, CH2, PA1,  TIM_USE_PWM,               0),
                                                                                                                                                                                                                                    ^
./src/main/target/NAZE/target.c:29:228: note: (near initialization for 'timerHardware[1]')
...
```
The warning does not seem to come from the number of elements (USABLE_TIMER_CHANNEL_COUNT) of the array, it seems to come from the number of elements of the structure of each element.

Each element has a value like this:
`DEF_TIM(TIM2, CH1, PA0, TIM_USE_PPM | TIM_USE_PWM, 0),`

This DEF_TIM is replaced when compiling, and by the error it seems that the latest parameter is being replaced for spaces or something similar, when LED_STRIP is undefined.

**HOW TO REPRODUCE**: comment the LED_STRIP feature in the `target/common.h` file and compile, al least if fails with the `make TARGET=NAZE`

This was first seen in https://github.com/cleanflight/cleanflight/issues/2654 but I think is better to have its own issue.

ledvinap, 2017-03-22:
this condition: https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/timer.h#L98-L106 must be replicated in cpu-specific DEF_TIM (https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/timer_def.h#L61 and others)

ledvinap, 2017-03-22:
Possibly by something like (untested):
```
#if defined(USE_DSHOT) || defined(LED_STRIP)
# define DEF_TIM_DMA_COND(...) __VA_ARGS__
#else
# define DEF_TIM_DMA_COND(...)
#endif

#define DEF_TIM(tim, chan, pin, flags, out) {\
    tim,\
    IO_TAG(pin),\
    EXPAND(DEF_CHAN_ ## chan),\
    flags,\
    (DEF_CHAN_ ## chan ## _OUTPUT | out),\
    EXPAND(GPIO_AF__ ## pin ## _ ## tim ## _ ## chan),\
    DEF_TIM_DMA_COND( \
       CONCAT(EXPAND(DEF_TIM_DMA__ ## tim ## _ ## chan), _CHANNEL),\
       CONCAT(EXPAND(DEF_TIM_DMA__ ## tim ## _ ## chan), _HANDLER)\
    } \
/**/
```

McGiverGim, 2017-03-24:
@ledvinap Sorry for my late reply, but I thinked you missed something in your code, I'm a Java developer and not a C developer ;)

I've tried your solution applying your sample code to the defines for F1, F3, F4 and F7 boards. I have tested that compiles for some of them, with and without `LED_STRIP` feature.

I've seen problems, for example with some of the F7 boards. For example the FURYF7 and the NUCLEOF7, they don't use the DEF_TIM macro in the `target.c`, they create the `timerHardware` fields directly, so if I remove the `LED_STRIP` feature I've the warnings and errors again.

It's not a problem, because only affects when the `LED_STRIP` feature is disabled, and the F7 boards not need to do it. If you want I can create a PR with your changes, and work from this to let the code as it must if you're interested in this. With some guide for one of them, I can verify and change it for all the targets.

Regards and thanks for all!

McGiverGim, 2017-03-24:
I've seen that the code for the `target.c` for the FURYF4 did not use the `DEF_TIM` but was modified to use it. I will take a look to see if I can understand the changes and apply it to the others.

McGiverGim, 2017-03-24:
Taking a look at all this codes, finally I've understood all the DEF_TIM stuff.

I will modify for ALL targets where DEF_TIM if missed and then I will make the PR.

Thanks for your help!

McGiverGim, 2017-03-25:
PR created https://github.com/cleanflight/cleanflight/issues/2677

Please, if you have some time, take a look to the code. 

Thanks!

hydra, 2017-03-29:
PR merged.

closed 2017-03-29 by hydra