cleanflight/cleanflight #2675: Beeper not working with default values in NAZE32 REV6

McGiverGim, 2017-03-23:
The beeper is not working in the NAZE32 REV6 with default values in CleanFlight 2.0. I've seen that the default values have this settings:
```
set beeper_inversion = OFF
set beeper_od = ON
```
In REV6 this settings must be changed to:
```
set beeper_inversion = ON
set beeper_od = OFF
```
I've seen this problem was detected some time ago here: https://github.com/betaflight/betaflight/issues/1396

**HOW TO REPRODUCE:** install a fresh version of CleanFlight, erasing all the data or in the config tab, press `Reset Settings`, you will see in CLI the default values and the beeper is not working.

McGiverGim, 2017-03-23:
This issue has a PR that solves it in Betaflight: https://github.com/betaflight/betaflight/issues/2734

I will close the issue when this PR is merged into Cleanflight.

Regards!

McGiverGim, 2017-04-02:
The PR has been merged into CleanFlight. I close the issue, now it works. Thanks!

closed 2017-04-02 by McGiverGim

luismi7777, 2017-05-02:
I have the opposite problem my  buzzer not stop to sound  a continuos beep and beep normally when activate mode flights  or beeper mode or armed  but with the continuos beeps always
Somebody knows how to fix this issue 

I mount other naze at the same time with the same buzzer and works perfect

mikeller, 2017-05-02:
@luismi7777: You can try to see if changing `beeper_inversion` or `beeper_od` settings fixes it. But if the boards are the same, running the same configuration, then it's most likely a hardware fault.

luismi7777, 2017-05-02:
One of them is a naze rev6 normal  and the other is dot 10 with mag and baro

And what have I to put in cLi
I'm newbie
Thanks for all

luismi7777, 2017-05-02:
@mikeller: can you say

Where are  this options for this commands
Or what have I write in cLi to change  this optios

luismi7777, 2017-05-02:
I try after reads

Set beeper_inversion= on/ off
Set beeper_od= on/off
Nothing change 

And the beeper always sound , is always on,   do it all the sound aux channel  and start  but always sound

mikeller, 2017-05-02:
Well, looks like it's a hardware fault then.

McGiverGim, 2017-05-03:
Or do you have the beeper mode on.