cleanflight/cleanflight #2678: Build Docs Inconsistency

biank88, 2017-03-26:
On `docs/development/build in Eclipse` it is stated that `Versions mandated below are current and correct as of January 2016'... however the the command `make arm-sdk-install' currently installs `gcc-arm-none-eabi-6-2017-q1-update` as opposed to the `The maximum, and currently recommended version, is 4.9-2015q3.` Is that docs are not uptodate or there's something I'm doing wrong?

McGiverGim, 2017-03-27:
All the docs for "building" have versions outdated. So you must use the version of the `make arm-sdk-install` and not the version of the docs.

Regards!

biank88, 2017-03-27:
I'd like to contribute to this project but I've found that the build documentation needs to be updated and that's slowing me down a lot... For example the `make` commands like `arm-sdk-install` or `arm-sdk-clean`  seem not to be documented at all.

Right now I'm trying to build cleanflight in ubuntu eclipse but the following error **section `.text' will not fit in region `FLASH'** has come up and I can't find a viable solution:
```
/home/fra/Desktop/cleanflight-workspace/[git-master]cleanflight/tools/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: obj/main/cleanflight_NAZE.elf section `.text' will not fit in region `FLASH'
/home/fra/Desktop/cleanflight-workspace/[git-master]cleanflight/tools/gcc-arm-none-eabi-6-2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 94728 bytes
collect2: error: ld returned 1 exit status
make[1]: *** [obj/main/cleanflight_NAZE.elf] Error 1
Makefile:1072: recipe for target 'obj/main/cleanflight_NAZE.elf' failed
make: *** [hex] Error 2
make[1]: Leaving directory '/home/fra/Desktop/cleanflight-workspace/[git-master]cleanflight'
Makefile:1167: recipe for target 'hex' failed

22:45:02 Build Finished (took 7s.248ms)
````

McGiverGim, 2017-03-27:
Remove the contents of the obj folder and try again.

biank88, 2017-03-27:
I've deleted the content of `obj` folder but the error is still there whenever I try to build again. 
Is there any other documentation online?

mikeller, 2017-03-27:
@biank88:

> `.text' will not fit in region `FLASH'
> `FLASH' overflowed by 94728

This tells you that you have too many features enabled in your configuration, which causes the binary to become too big to fit into the flash space of your MCU.

(In reality, enabling _any_ more features for NAZE will most likely cause this, since there is very limited flash space available for this target.)

McGiverGim, 2017-03-28:
Like @mikeller said, if you activate something without removing other features, you will get this error.

I've a PR with a document explaining how to enable or disable features, you can take a look to it in:
https://github.com/McGiverGim/cleanflight/blob/document-customized-version/docs/Customized%20Version.md

biank88, 2017-03-28:
Thanks @mikeller, @McGiverGim. I hope that PR gets merged soon.
I've finally succeded in building `cleanflight@master` for `TARGET=SPRACINGF3 DEBUG=GDB` by removing the following features:

#define USE_CLI
#define USE_PPM
#define USE_PWM
#define SERIAL_RX
**//#define USE_SERIALRX_CRSF       // Team Black Sheep Crossfire protocol
//#define USE_SERIALRX_IBUS       // FlySky and Turnigy receivers
//#define USE_SERIALRX_SBUS       // Frsky and Futaba receivers
//#define USE_SERIALRX_SPEKTRUM   // SRXL, DSM2 and DSMX protocol
//#define USE_SERIALRX_SUMD       // Graupner Hott protocol
//#define USE_SERIALRX_SUMH       // Graupner legacy protocol
//#define USE_SERIALRX_XBUS       // JR**

#if (FLASH_SIZE > 64)
#define BLACKBOX
#define LED_STRIP
**//#define TELEMETRY
//#define TELEMETRY_FRSKY
//#define TELEMETRY_HOTT
//#define TELEMETRY_LTM
//#define TELEMETRY_SMARTPORT**
#define USE_RESOURCE_MGMT
#define USE_SERVOS
#endif

#if (FLASH_SIZE > 128)
#define GPS
#define CMS
#define TELEMETRY_CRSF
**//#define TELEMETRY_IBUS
//#define TELEMETRY_JETIEXBUS
//#define TELEMETRY_MAVLINK
//#define TELEMETRY_SRXL
//#define USE_DASHBOARD**
#define USE_MSP_DISPLAYPORT
#define USE_RX_MSP
#define USE_SERIALRX_JETIEXBUS
#define USE_SENSOR_NAMES
**//#define VTX_COMMON
//#define VTX_CONTROL
//#define VTX_SMARTAUDIO
//#define VTX_TRAMP**
#endif

---

However there's something I don't understand in the build process. 

In particular, I assumed that once a target has been defined in the eclipse project properties, for example `TARGET=SPRACINGF3 DEBUG=GDB`, than there's no need to customize `target/common.h` in order to get a successful *default build* for that specific target.

Also, what `target/common.h` file is used to build the firmware releases available in the Cleanflight configurator?

mikeller, 2017-03-28:
Ah I see. Yes, you do not have to change the source files in order to be able to build a target as distributed.
In your case, what caused the size overflow was `DEBUG=GDB`. Building a debug target increases the size of the binary considerably, since it turns off all optimisations, size and otherwise. Unfortunately, this means that debug builds won't work for F1 targets, and probably neither for a lot of the F3 targets.

Re your comments above, you should never have to modify `target/common.h` in order to be able to build a single target. What you should do (if you have to modify anyting), is to modify the target specific file in `target/<target name/target.h` for example if you want to stop telemetry from being built, and there is a `#define TELEMETRY` in `target/common.h`, simply add an `#undef TELEMETRY` into the `target.h` file for your target. Since this file is parsed after `common.h`, it will undo the earlier define.

McGiverGim, 2017-03-28:
@mikeller thanks for your info about `common.h`, I've created a document about create a customized version (there's a PR about this), and it's the doc that @biank88 has followed I think.

I will modify the doc with your info about the best way to do it.

Thanks!

biank88, 2017-03-28:
Thanks for your support!
BTW I'm experiment with MDwiki engine in order to organize some docs I need: [example](https://biank88.github.io/cleanflight-wiki/en/#!pages/dev/building-in-eclipse.md)
Do you think it would be useful?

hydra, 2017-04-17:
documentation PR merged, closing.

closed 2017-04-17 by hydra