cleanflight/cleanflight #2682: BLHeliSuite passthrough not working with CleanFlight 2.x

McGiverGim, 2017-03-30:
I can't connect to the ESC from BLHeliSuite version 2.x using the CleanFlight passthrough, it appears this message:

[image]

If I flash CleanFlight 1.4.2 then I can connect without problem.

My setup is a NAZE32 REV6 10DOF with Littlebee fvt Opto Pro 20A ESC.

hydra, 2017-03-30:
@4712 can you have a look into this?

McGiverGim, 2017-03-31:
I've tested a 2.x version from some days ago and the BLHeliSuite passthrough worked again. So some of the recents commits have breaked it.

McGiverGim, 2017-03-31:
@hydra @4712 I think I've found the commit who introduce the problem:
https://github.com/cleanflight/cleanflight/commit/ab2b1188cfeedd38a7781dad93ae1902b0736f09

Maybe some of the commands cleaned is used by BLHeliSuite?

Regards!

4712, 2017-03-31:
@McGiverGim @hydra: removing MSP_MISC introduced it... I need to switch to MSP_MOTOR_CONFIG I guess.

hydra, 2017-03-31:
@4712 yes that's likely.  we're removing all the old crufty MSP commands from CF (and BF).  Likely the change in the BlHeliSuite should be simple enough.  Are you able to do it?

4712, 2017-03-31:
@hydra Sure, asap.

McGiverGim, 2017-03-31:
@4712 I will test it as soon you have the first version ;)

Regards!

4712, 2017-04-01:
Fixed in [BLHeliSuite 16..6.14.9.0.1](https://www.mediafire.com/folder/dx6kfaasyo24l/BLHeliSuite)

McGiverGim, 2017-04-01:
@4712 Tested and working ;)

I close the issue.

Thanks!

closed 2017-04-01 by McGiverGim