cleanflight/cleanflight #2685: Link error if deactivate PWM to free some space

McGiverGim, 2017-03-31:
Any of the features of CleanFlight can be deactivated to get some free space to add others.

I'm playing with it and the NAZE target.

If I add to the target.h of NAZE this:
```
#undef USE_PPM
#undef USE_PWM
```
The hex is created without problem, but if I add only:
```
#undef USE_PWM
```
The linker gives me this error:
```
Linking NAZE
/tmp/ccCHoCnV.ltrans1.ltrans.o: In function `servoDevInit':
/mnt/d/git/cleanflight-2.x/./src/main/drivers/pwm_output.c:354: undefined reference to `pwmConfig_System'
collect2: error: ld returned 1 exit status
make[1]: *** [obj/main/cleanflight_NAZE.elf] Error 1
make[1]: se sale del directorio «/mnt/d/git/cleanflight-2.x»
make: *** [hex] Error 2
```
The error can point to different .c files depending of what features I've enabled or disabled, but only happens when I activate PPM and not PWM. Maybe some code of PPM references PWM.

hydra, 2017-03-31:
yeah, search the code for references to `pwmConfig` and see where it is used without surrounding #ifdef USE_PWM to find it.

McGiverGim, 2017-03-31:
Ok, this is a task I can do :)

McGiverGim, 2017-03-31:
Problem found. I will close the issue when the PR will be merged.

Thanks!

hydra, 2017-04-01:
pr merged, thanks.

closed 2017-04-01 by hydra