cleanflight/cleanflight #2694: CleanFlight 2.x + GPS enabled in NAZE32: randomly disconnects

McGiverGim, 2017-04-02:
Hi! I don't know if you can help me with this, because it's a customized version of firmware, but I think that the code must work.

I know that GPS is deactivated for F1 targets. I've made my own personal compilation of CleanFlight 2.x for my NAZE32 deactivating a lot of features and activating the GPS in SoftSerial 1 at 9600. I've activated too the MAG and other sensors.

The GPS is detected without problem, my CPU is over 20% and the GPS only consuming 0.8% of this CPU.

The problem is that the GPS appears and disappears after few seconds. I don't know why. Making it totally unusable.  In CleanFlight 1.4 it worked perfectly, but I don't want to return to this version because with 2.x the quadcopter flies better.

The only thing strange that I see is that the CPU in `tasks` is far higher than in `status` I don't know if it's normal, with the serial and gyro very high:
```
# status
System Uptime: 20 seconds
Voltage: 42 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=72MHz
Stack size: 2048, Stack address: 0x20005000
I2C Errors: 0, config size: 1222, max available config: 2048
CPU:21%, cycle time: 1010, GYRO rate: 990, RX rate: 48, System rate: 9

# tasks
00 - (         SYSTEM)      9       9       2    0.5%    0.5%         3
01 - (       GYRO/PID)    946    1410     799  133.8%   76.0%    105819
02 - (          ACCEL)    964     332     190   32.5%   18.8%     26121
03 - (       ATTITUDE)     49     870     499    4.7%    2.9%      5262
04 - (             RX)     47     276     124    1.7%    1.0%       817
05 - (         SERIAL)     24  472986       5 1135.6%    0.5%       549
07 - (BATTERY_VOLTAGE)     24      62      20    0.6%    0.5%       105
09 - ( BATTERY_ALERTS)      2      15       4    0.5%    0.5%         2
10 - (         BEEPER)     24      18       3    0.5%    0.5%        26
11 - (            GPS)      4     862     366    0.8%    0.6%       290
12 - (        COMPASS)      2     141     131    0.5%    0.5%       163
13 - (           BARO)     10     161      90    0.6%    0.5%       465
14 - (       ALTITUDE)      9     372     233    0.8%    0.7%       889
15 - (      TELEMETRY)     58      26       4    0.6%    0.5%        87
RX Check Function                64       9                        55
Total (excluding SERIAL)                     178.1%  103.5%
```
All is working without problem, only GPS seems randomly loosing the connection.

Thanks!

McGiverGim, 2017-04-02:
I forgot to say that if I deactivate `Accelerometer` and `Barometer` the CPU is only at 4% but the problem ramains.

McGiverGim, 2017-04-02:
I've tried to lower the `Gyro update frecuency` and `PID loop frecuency` with all the sensor active to test, and my CPU is under 5% but the GPS disconnects as before.

McGiverGim, 2017-04-02:
I've seen a comment, from more than a year ago (when the GPS was disabled for the NAZE32) of other person with the same problem:
https://github.com/betaflight/betaflight/issues/57#issuecomment-166654063

So maybe this is an old bug.

hydra, 2017-04-02:
Recommend upgrading to an F3 board.

Softserial is ok for transmissions (i.e. telemetry) but other system load causes it to suffer when receiving data, even at 9600.  Likely you'd need to log and inspect the actual data received by the FC and the data sent by the GPS and compare the differences.  You'd probably need a scope and some 'pin debugging' to find out the exact timing issue.

I certainly have no time for this time-consuming process when the world has moved on to F3 or better processors which handle your scenario much better.

I should note, that while softserial is useful for some limited cases it has always had issues when people try and use it for high speed or high bandwith scenarios.

Are you sure your GPS unit isn't sending out loads of unneccessary messages?  is it correctly configured.

This sounds like more of a question for a user-forum kind of issue rather than a github issue.

hydra, 2017-04-02:
I will close this issue, there is not enough information and it's unclear under exactly what conditions the problem occurs.  we'd need to know things like:
1) exact configuration of FC (cli dump).
2) exact configuration of GPS unit (steps to reconfigure an identical one).
3) make/model/firmware of GPS unit.
4) wiring configuration.

if you can provide a clearer bug report please open a new issue and link to this comment.

closed 2017-04-02 by hydra

McGiverGim, 2017-04-02:
Ok, I will stay in version 1.x until I bought an F4. I've seen that some F3 had lose the GPS support too.
I can give you more information but it will be more work for you and I'm the only with this problem 😉
I opened this only to see if someone have an idea to test.
Thanks!