cleanflight/cleanflight #2696: Prepare for CF 2.0.0 release

hydra, 2017-04-02:
CF 2.0.0-RC2 is out now.  Please leave bug reports/testing feedback as comments below.

nightflasher, 2017-04-03:
uploaded to my naze32 rev6 10dof => nothing...absolute nothing is working.
even tried to compile by myself to get BARO and Sonar working, but it's also not working. even not compiling with all sensors enabled.

mikeller, 2017-04-03:
> absolute nothing is working

So your flight controller does not even connect in the configurator? Try reflashing with 'Full chip erase' enabled.

In order to be able to build with BARO and SONAR, you will have to disable some other features to make it fit into the small amount of flash available on F1 chips. Unfortunately, there is not much left to be disabled on F1 targets like the NAZE.

nightflasher, 2017-04-03:
i always  use "Full chip erase", so that's not the problem. after flashing and dis-/reconnecting power and usb the configurator only shows the version of the board (right board) but nothing more. i already have tested the v2.0.0 of cleanflight configurator, so it can be eliminated as a faulty point.

sooo...what's next?

mikeller, 2017-04-03:
Will test this with a NAZE later on when at home.

McGiverGim, 2017-04-04:
I've tested my NAZE32 REV6 10DOF a lot this days, including the actual code in the repository and it works perfectly. I've added MAG and BARO and it's working. I haven't tried with SONAR because I didn't have one.

tteskac, 2017-04-04:
I can't get BARO to work neither, but on CC3D revC. 
Is it normal that in the "resource list" for B10 and B11 (UART3) stands "FREE" but not "I2C_SCL/SDA"?

And for SERVO_TILT I had to remap resource SERVO 1 -> MOTOR 5 and SERVO 2 -> MOTOR 6.

nightflasher, 2017-04-04:
i've tried again, my compiled version and the one in unstable in the cf configurator, it's always the same. 
my boards are "afroflight32", maybe this is a different one to  @McGiverGim !?
https://anonimag.es/image/JT9snEH

McGiverGim, 2017-04-04:
Maybe it's a stupid question... Have you activated it into the configuration tab?

nightflasher, 2017-04-04:
@McGiverGim as you can see on the picture, i can't do anything. not in the "old" configurator nor the "new" one. I'm stuck on the first tab AND i even can't tick the "Reset Settings" button.

mikeller, 2017-04-04:
@nightflasher: I couldn't download the picture. But if it's not an original NAZE (which is what I've got) then there is not much point in me verifying that it works on different hardware. Sorry I couldn't help.

McGiverGim, 2017-04-04:
If I'm not wrong, this happened to me when using the master branch of the cleanflight-configurator.
Please, use this: https://github.com/cleanflight/cleanflight-configurator/commits/update-for-api-1.33.0 to see if it solves the problem.

nightflasher, 2017-04-04:
@McGiverGim you're best bro! :-D now i can flash AND use my firmware...now i'm testing

nightflasher, 2017-04-05:
so now, first flights in my sleeping room (14m² ; who also tries first flights in the sleeping room?) seems very good to me. thumbs up! hope there's some time to fly outside. a friend of me will test with hexacopter, i'm testing on my ZMR 250.

btw: where to put pid-config and setup examples? i think this could be a good startpoint for newbies ;-)

hydra, 2017-04-06:
@nightflasher thanks for feedback - release notes of CF2.0 had the links to the the configurator BTW 😄

McGiverGim, 2017-04-08:
I've made several tests today with the new code. I've seen this commit: 
https://github.com/cleanflight/cleanflight/commit/99c87ba2dc930ad2e426f69f7fd19eff2ed63e98
changed the defaults so I suppose some code has changed in the PID controller.

With the code of one week ago, with the default values, my quad was totally flyable. But with the code of today and default values, every throttle punch mades my quad derive to one side (sometimes left, sometimes right). I see in the blackbox that the error is trying to be collected by the I gain, but it is too slow. 

I've seen too that the quad does not hold the desired angle, so for me it's clear that I need to increase the I gain. This happened too with the code of one week ago.

Maybe the new code makes the quad more sensible to thottle punch when the I gain is low? Or this happens only to my quad?

Regards!

rgn15996, 2017-04-08:
Flashed 2.0RC2 to my SP3NEO+VTX with full chip erase. OSD Graphic elements (e.g. battery and Signal strength icons) are corrupt. Alphanumerics all OK - but there is no option in the configurator to upload a font as there is in BF - unless I am missing something?

mikeller, 2017-04-09:
Yes there is. Bottom right, see image.

[image]

rgn15996, 2017-04-09:
Ah - I don't see that in the version of the configurator I am using - just a save button as below. 
[image]

I was using the directed in the release notes (cleanflight-configurator-update-for-api-1.33.0

rgn15996, 2017-04-09:
OK - I just looked with Chrome tools and the button is hidden with an applied style
[image]

If I unhide it and click on the button, I can load up the font! Presumably some logic is incorrectly hiding this from my SP3NEO+VTX?

hydra, 2017-04-09:
@rgn15996 thanks for the report, I'll look into that for you.

hydra, 2017-04-09:
just checked on my NEO with the latest build of master and the latest configurator from the new branch and it's OK for me, can you confirm this?

wjreeds, 2017-04-10:
Even though I am new to Cleanflight, I decided to give v2.0.0 RC2 a try; once I figured out I needed to use the developmental configurator I was able to connect no problem.  The issue I have ran into is that my Mag is not recognized in this version.  It was recognized and I was able to use it using the last stable version.

I am using an EMAX Skyline32 Advanced V.2 (which uses the NAZE firmware in Cleanflight)
[image]
[image]

McGiverGim, 2017-04-10:
@wjreeds The MAG was removed in NAZE32, but it has been enabled again three days ago: https://github.com/cleanflight/cleanflight/commit/0cf08ec8c042f679731e178ec02e3900b5a1e15e

So you can compile the actual master code or wait for the next version published.

rgn15996, 2017-04-10:
@hydra So I should try with the latest master cleanflight and the latest main branch of configurator (rather than the API version1.33.0?

asulwer, 2017-04-10:
I flashed to spracingf3neo (with osd attached) from rc1 to rc2 and none of the menu config options will work.  I had to roll back to 1.14 then to rc1 before I was back to what I started with.

When I click on any of the menu options the config app doesn't go to that section.  I am stuck on the menu that has the Calibrate Accelerometer button.

mikeller, 2017-04-10:
@asulwer: Try this branch of the configurator:  https://github.com/cleanflight/cleanflight-configurator/tree/update-for-api-1.33.0

hydra, 2017-04-11:
@asulwer use latest master and the latest version of the configurator from here:
https://github.com/cleanflight/cleanflight-configurator/tree/update-for-api-1.33.0

Angrybee23, 2017-04-14:
MOTOR_STOP isn't working.  I'm running CLFL version 2.0.0  MultiWii API version 1.35.0 Flight Controller is a SPRACINGF3 and the ESC's are Makerfire 20 amp with OneShot125. I reflashed with full chip erase and it's still not working.  The motors start spinning when I arm the quad and the RPMS slowly get higher over time.  If I drop the Minimum Throttle from the default 1070 to 1000 the motors don't spin once armed until I give it some throttle but when moving the throttle back down the stay spinning and slowly and the RPM's start to go up again...  This didn't happen with firmware 1.14.2 and the older version of the configurator.

Angrybee23, 2017-04-14:
Sorry, I had airmode enabled lol.

hydra, 2017-04-14:
@Angrybee23 thanks for the update.  closing this issue now,

hydra, 2017-04-14:
For bug reports on CF v2.0.0 please add comments to https://github.com/cleanflight/cleanflight/issues/2720 for the CF v2.0.1 release.

closed 2017-04-14 by hydra