cleanflight/cleanflight #2698: GPS no stable with CleanFlight 2.x (working again with some changes)

McGiverGim, 2017-04-03:
@hydra I open this issue related to this: https://github.com/cleanflight/cleanflight/issues/2694 because I've new info. I don't know if this info that I will put here is what you need, if not, please close this issue again.

As said in the other issue, the GPS connects and disconnects each few seconds, in a NAZE32 customized compilation to activate the GPS.

In CleanFlight 1.x it was stable as a rock, and I'm not a C developer, son my knowledge is limited, but I've maked some tests, and one easy to test is to replace the `gps.h` and `gps.c` from CleanFlight 1.x into CleanFlight 2.x (and adding some other corrections to references as needed)... ¡¡¡and it works again!!! Now is solid stable as rock again. So the problem is in the code of GPS and not in the communications or some other stuff.

You can see the commit that make the magic here're, not too much changes:
https://github.com/McGiverGim/cleanflight/commit/abd6634b74cf719f76888ad6ed3c65413ec14df5

You can see the changes. I needed to add the `fc_serial.c` file that surely is now in other place in the code, so my commit needs to be corrected by sure.

Is the code based in BetaFlight better than the old CleanFlight code? Now seeing the changes, maybe you have an idea of what can be wrong, or at least someone can give me some advice to let the code better and easier to maintain (I will use it and I want to merge into it futures releases of CF 2x.).

My configuration, if someone want to reproduce the problem in a NAZE32:
- CleanFlight 2.x with this customizations to enable the GPS into ```target.h```:
```
// Customized defines
#undef USE_PPM
#undef USE_PWM
#undef USE_SERIALRX_CRSF       // Team Black Sheep Crossfire protocol
#undef USE_SERIALRX_SBUS       // Frsky and Futaba receivers
#undef USE_SERIALRX_SPEKTRUM   // SRXL, DSM2 and DSMX protocol
#undef USE_SERIALRX_SUMD       // Graupner Hott protocol
#undef USE_SERIALRX_SUMH       // Graupner legacy protocol
#undef USE_SERIALRX_XBUS       // JR

#undef LED_STRIP
#undef TELEMETRY_FRSKY
#undef TELEMETRY_HOTT
#undef TELEMETRY_LTM
#undef TELEMETRY_SMARTPORT
#undef USE_SERVOS

#define GPS
```
- Uncomment the MAG into the `target.h`
- Diff command:
```
# diff
# Cleanflight / NAZE 2.0.0 Apr  3 2017 / 10:00:59 ()
name Robocat
resource SERIAL_TX 10 A00
feature -RX_PPM
feature VBAT
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY
serial 0 1 19200 57600 0 115200
serial 1 4160 115200 57600 115200 115200
serial 30 2 115200 9600 0 115200
aux 0 0 0 1950 2050
aux 1 27 1 1750 1850
aux 2 1 1 1050 1450
aux 3 1 1 1550 1650
aux 4 2 1 1450 1550
aux 5 5 1 1550 1650
aux 6 5 1 1150 1250
aux 7 3 1 1050 1450
aux 8 3 1 1550 1650
aux 9 4 1 1050 1450
aux 10 4 1 1550 1650
aux 11 9 1 1250 1350
aux 12 10 1 1050 1250
aux 13 12 1 1250 1350
aux 14 12 1 1850 1950
aux 15 25 3 1950 2050
rxfail 5 s 1200
set gyro_notch2_cutoff = 150
set acc_trim_pitch = 4
set acc_trim_roll = 2
set mag_hardware = AUTO
set baro_hardware = AUTO
set min_check = 1050
set max_check = 1950
set fpv_mix_degrees = 15
set serialrx_provider = IBUS
set blackbox_rate_denom = 2
set min_throttle = 1052
set failsafe_throttle = 1350
set failsafe_procedure = AUTO-LAND
set vbat_max_cell_voltage = 42
set current_meter = NONE
set disarm_kill_switch = OFF
set gps_provider = UBLOX
set gps_auto_baud = ON
set deadband = 3
profile 0

set d_pitch = 23
set d_roll = 24
rateprofile 0

set rc_expo = 25
set rc_yaw_expo = 25
set thr_expo = 55
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
```
- I'm using this knwon GPS: http://www.banggood.com/NZ-GPS-For-NAZE32-Flip32-6dof-10dof-Best-For-QAV250-ZMR250-Multicopter-Quadcopter-p-1015134.html?rmmds=search
- Connected to the SoftSerial 1 of NAZE32.

HOW TO REPRODUCE: enter into the configurator. The GPS will connect and the icon will be green. Some seconds after that (can be 20 seconds or 2 minutes, but no more), it will disconnect and the icon will be black. Some seconds after that it will connect again, etc.

Thanks in advance to all!

McGiverGim, 2017-04-03:
I've updated the commit removing some things added that were not necessary. I will try to add/remove some of the changes to try to refine the part that makes it fail. I will post my findings here.

McGiverGim, 2017-04-04:
Found the line that makes if fail, opened PR https://github.com/cleanflight/cleanflight/pull/2699

hydra, 2017-04-06:
good work!  thanks for digging into it, PR merged.

closed 2017-04-06 by hydra