cleanflight/cleanflight #2703: Request: Make sure 3D-mode is SAFE before taking over betaflight code

fftunes, 2017-04-07:
Because as it is now, it is not safe. Theoretically, motors could be commanded to 100% while/before reversing because of unrestricted PID activity. Numerous users in Boris' blog on rcg reported problems with switching direction. If you blindly take over betaflight 3D code, broken quads are pretty much guaranteed, especially for those using older, less powerful escs & motors.

I kept suggesting different solutions... the most simple attempt would be to:
- add a configurable delay during which PID are zeroed before reversing motors, to allow motors to idle down before attempting to reverse. Delay can be configured to match your gear and flying style.

I got a few 3D quads for testing if somebody is willing to work on code.

Greetings & thx in advance, 
ff

hydra, 2017-04-09:
Hi.  Please provide steps to repeat the problem and/or blackbox logs and I'm sure one ofthe betaflight PID/mixer gurus can take a look, without this information it might be a bit hard to get someone to start working on it.

borisbstyle, 2017-04-10:
@fftunes what fc commands to esc doesnt matter at all. It is internal ESC firmware that is responsible to provide protection against destroying itself.
When someone wants to fly 3D he should  make sure that the gear uses is suitable for that.
Even the most modern ESCs are still struggling with it so I really dont suggest to use old escs.
The type of props for 3D flying should also be as liggt as possible to improve reliability.

Latest version also has possibility to disable airmode fully. That will soften the transition btw.

hydra, 2017-04-11:
thanks for the reply @borisbstyle closing the issue.

closed 2017-04-11 by hydra

fftunes, 2017-04-11:
lol @hydra it's simple physics that it's not a good idea to try instant reversing when some of the motors might be commanded really high, how is that not-enough-information. Could at least have had discussion with more than 1 guy who doesn't even fly 3D hint hint. Any little bit of "brake time" would help a lot for a in-sync reversing... how can i put it simpler than that.

martinbudden, 2017-04-11:
@fftunes , 

> it's simple physics that it's not a good idea to try instant reversing when some of the motors might be commanded really high

Of course. However the argument is that the ESC should handle this. So if the output to the ESC changes from 100% forward to 100% reverse, it is the ESC that handles the spinning down and reversing of the motors. The ESC has information that the flight controller doesn't, like the motor RPM and so can handle this transition better.

borisbstyle, 2017-04-11:
@fftunes Physics is up to the ESC to determine if it is possible or not. The actual ESC code has slewing implemented. Same thing also goes when going 0 to 100% throttle instantly. No ESC can handle that if it was really that way. The ESC firmware internally only allows certain amount of steps at the time.
FC is controlling the ESC's desired throttle output, but it is the ESC that controls the actual RPM.

But I also mentioned above. You can now disable airmode, which will remove about 50% of stabilisation around transition point with 3D. Pretty much exactly the same like old cleanflight code before airmode.

mikeller, 2017-04-11:
This discussion reminds me of the guys slamming their cars into reverse while still rolling, and then wondering when their clutch and gear box is wrecked after 10000 km...

fftunes, 2017-04-21:
@martinbudden But the escs are not the problem here, they do their job just fine. It's more of a problem what the FC starts doing when the motors reverse at different times (think of yaw momentum correction for example).

@borisbstyle Yes i seen that airmode can be completely disabled and i agree that this might help a lot, but imho that's not a real solution when i ask to make 3D mode more "safe", no matter the modes, besides that most people would like to run both airmode & 3D at the same time.

@mikeller I'm sorry if i got you wrong, i'm not sure what you're trying to say. In my understanding "the guys" in this case would be the FC i.e. clean-/betaflight 3D code commanding the escs to do stupid things.

Let me point out again that as long as the escs don't talk to each other, it's the FC's responsibility to help them reversing in sync.

Now if there's simply no interest at all to improve 3D mode, that's a different story.

mikeller, 2017-04-28:
@fftunes: I was just trolling for once. I am sorry, I should have added a :-P to denote irony.

I think you make a valid point there, the way that 3D works for now makes it entirely possible for the motor outputs of the flight controller to be in a state where some are sending CW, and others CCW to the ESC. This is probably something that should be constrained, considering that the changeover from one direction to the other will introduce a considerable period of little or no thrust for every motor.