cleanflight/cleanflight #2705: No sticks input with IBUS on NAZE since BF build #1168

AirBreak69, 2017-04-08:
Using current build on a good old NAZE with IBUS the stick inputs stuck at 1500.
Switches and VRs are still working.
Problem was introduced in BF build 1168. see https://github.com/betaflight/betaflight/issues/2838
Cleanflight catched it with the betaflight merge from 2017/04/06.
Another problem with this and earlier builds: AUX2 is copied to AUX6, AUX6 does not work.

hydra, 2017-04-11:
@AirBreak69 thanks for the report.  I'll keep an eye on that bug report.

hydra, 2017-04-11:
@McGiverGim can you test latest master and confirm is there is still an issue please?

McGiverGim, 2017-04-11:
Tested. In my setup it works perfectly. Used actual code in master.

My setup is a NAZE32 REV6 10DOF, with a FlySky FS-i6S and a FlySky FS-IA10B, sharing ibus and telemetry in the same UART2.

hydra, 2017-04-11:
thanks for report, closing as fixed.

closed 2017-04-11 by hydra

AirBreak69, 2017-04-11:
Tested. The sticks values are back again.
The mentioned AUX2 to AUX6 copy was a faulty setup in my transmitter - sorry.
But: AUX5 initializes with a value of 1900 if the transmitter is not active on start up.