cleanflight/cleanflight #2706: CC3D, PWM and motor spin

JotaStar, 2017-04-08:
Hi,

My quad configuration is an old CC3D with FS-R6B Receiver that have 6Ch via PWM.

From the first moment I had problems with the receiver. The channels were changed and some did not arrive. After much Googling, it was a problem with looptime. The configuration of the radio / transmitter is standard mode 2 (AETR1234) however I had to set it to ETRA1234 to get all channels well. I do not quite understand why but at least it works.

Now the problem I have is when trying to arm the motors not all do. Only No.4 is armed and it works. I had playing with the looptime to make the motors arm, but then the radio is not going well and vice versa. And if everything works at time, then the motors start spinning randomly.

Does anyone make it run with a similar setup without problems?

What is more curious is that with LibrePilot everything works well. The engines are armed, the radio works and the channels are in the order they should ... And I have seen in the LibrePilot configuration that its looptime is 2000ms (0.5Hz) and ESC at 490Hz. The ESCs are OneShot125, functionality that I DO NOT use since I think with PWM it is not possible.

What else can I try? Any ideas?

Thank you in advance

hydra, 2017-04-09:
it's doubtful that there's in issue with the code since you've tried 3 different firmware versions.  more likely it's a configuration error.    check the painless360 youtube channel for setup guides and explanations.  use rcgroups.com forum threads for supports.  if you've found a bug please report it on a new issue here and link to this from it please.

closed 2017-04-09 by hydra

longxk, 2017-05-18:
@JotaStar Hi, I have exactly the same problem, how do you solve it?

longxk, 2017-05-19:
@hydra see this betaflight [issue](https://github.com/betaflight/betaflight/issues/2177), and my [comment](https://github.com/betaflight/betaflight/issues/2177#issuecomment-302600450), I think the firmware does have some problems.

JotaStar, 2017-05-19:
@longxk so I asume that is firmware issue and de have ti wait, isnt it? 
Or buy a PPM receiver...

thomaskilian, 2017-10-26:
Just to chip in: I have exactly the same issue here. (Well, I ordered a F3 meanwhile, but being able to use this scratch CC3D would be nice.)