cleanflight/cleanflight #2708: Wrong scale in blackbox data?

McGiverGim, 2017-04-09:
I don't know if this is a "problem" of Cleanfligth storing incorrect data in the BlackBox, or from the BlackBox log viewer.
To tune the PID I use some graphs, I like to have in one of them:
- The RC command (what I must that the quad do)
- The gyro info (what is doing the quad)
- The PID Error (difference between the others two)
My objective is try to let the PID Error as near as zero as possible.

There's a "problem", for example, in a roll movement.

In CleanFlight 1.x:
[image]

As you can see, the RC command and Gyro have similar values (they have different scales, but the legend show the value and is similar). The PID Error can show a little error at the beginning (intertia) but it's nearly zero all the time. All is good.

In CleanFlight 2.x:
[image]

Now, the RC command and the Gyro seems to have different scales. The Gyro never arrives to the values of RC. The behavior is like it has arrive. The PID Error shows the difference between both, and is constant, more or less.

Maybe some of the values is missing a rate or similar? It's an error or a "feature" of the new version?

hydra, 2017-04-09:
did you use the BetaFlight black box log viewer for CF2.x?  If not please test that.  If it's the same let us know.

hydra, 2017-04-09:
also, it's probably a new feature - you'd have to check with @borisbstyle or @martinbudden regarding the PID changes.

McGiverGim, 2017-04-10:
Yes, it's with BetaFlight Blackbox Explorer. I always used it.

fftunes, 2017-04-11:
@McGiverGim doesn't it look like just loose PID or something... try increasing dterm transition slider.

McGiverGim, 2017-04-11:
@fftunes It does not seem a problem of PID. I've been playing with it and the quad flies perfectly.

Can you try to show the PID_ERROR in some of your blackbox data to be sure? Thanks!

McGiverGim, 2017-04-12:
After studying the logs, it seems to me that the info in the BlackBox logs is correct. I've searched for a 360 roll and the rcCommand shows 675 deg/s (the maximum configured by me in PID tab), but the gyro is rolling at less velocity (between 300 and 500 deg/s) . It lasts one second, more or less to finish the roll, so it seems that the measures are accurate.

I use AirMode, and the motors are about 30% while rolling.

The problem appears in more slowly movements too, as the capture posted, so my rolling speed seems to be always under the desired command. Can someone post a BlackBox logged with the latest version to see if this is only a problem of mine or not?

McGiverGim, 2017-04-12:
I've found a blackbox log of other person with BetaFlight 3.1.6 and the graphs are correct, so or is a problem of my setup or is a problem in CF 2.0/BF 3.1.7.

I will continue looking for logs with the latest version.

McGiverGim, 2017-04-12:
Finally I found a BlackBox log with 3.1.7 of other person and it does not have this problem. So it's clear that is a problem of my setup as @fftunes said. I don't know why exactly, but I'll look into support in other places, as this is not an issue. I close it.

Thanks to all!

McGiverGim, 2017-04-16:
Finally this was really an issue, PR made https://github.com/betaflight/blackbox-log-viewer/pull/56

hydra, 2017-04-17:
@McGiverGim thansk for the update.

closed 2017-04-17 by hydra