cleanflight/cleanflight #2714: GPS and serialpassthrough BUG!

denisn73, 2017-04-13:
- The new version does not save the protocol of UBLOX GPS.
- Through serialpassthrough RX data recieved, but in the U-center module is not determined (with COM-adapter is OK)
Before the upgrade everything worked!

McGiverGim, 2017-04-13:
What FC do you have?

denisn73, 2017-04-13:
NAZE32 rev.6. Tested yesterday, it had upgraded today and now not working

McGiverGim, 2017-04-13:
GPS is disabled for NAZE32 in CF2.x. Have you build your own firmware?

stanislavdimov, 2017-04-13:
Where write that GPS for NAZE32 on CF 2.0 is disable???

McGiverGim, 2017-04-13:
In the code, or in betaflight docs. You can create your own version activating it https://github.com/cleanflight/cleanflight/blob/master/docs/Customized%20Version.md

stanislavdimov, 2017-04-13:
To be flight with CF on copter you must study programming!

McGiverGim, 2017-04-13:
Yes, if you aren't happy with what developers do... for free... 
If you don't know how to do it, you can rest in version 1, change to inav or change your NAZE. You decide.

denisn73, 2017-04-14:
McGiverGim, Yes, I can build the firmware. Where in the code to activate U-Blox and make it by default?

stanislavdimov, 2017-04-14:
If  folow instruction when open Cygwin-Terminal and try make hex:
make TARGET=NAZE
see :
-........
 %% bus_i2c_stm32f10x.c
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
%% dma.c 
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
%% light_ws2811strip_stdperiph.c ateProcess: No such file or directory
make[1]: *** [Makefile 1240: obj/main/NAZE.........stm....]Error 1ar
make[1]:***Waiting for unfinished job.....No such file or directory
make[1]: *** [Makefile 1240: obj/main/NAZE.........stm...].   Error 1
make[1]: *** [Makefile 1264: obj/main/NAZE.........startup.....] Error 1
make[1]: *** [Makefile 1240: obj/main/NAZE.........bus...._   ] Error 1
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
%%serial_uart_stm32f10x.c
make[1]: *** [Makefile 1240: obj/main/NAZE.........gpio...._   ] Error 1
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
make[1]: *** [Makefile 1240: obj/main/NAZE.........dma.o...._   ] Error 1
make[1]: *** [Makefile 1240: obj/main/NAZE.........inverter..._   ] Error 1
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
make[1]: *** [Makefile 1240: obj/main/NAZE.........light..._   ] Error 1
arm-none-eabi-gcc.exe: error: CreateProcess: No such file or directory
make[1]: *** [Makefile 1240: obj/main/NAZE.........serial..._   ] Error 1
make[1]: *** Leaving directory ' ....'
make[1]: *** [Makefile 1344: hex] Error 2

McGiverGim, 2017-04-14:
@denisn73 you don't need to activate ublox, you need simply to activate GPS, and then the ublox will work again https://github.com/cleanflight/cleanflight/blob/master/docs/Customized%20Version.md

McGiverGim, 2017-04-14:
@stanislavdimov It seems that some files are missing or that the compiler does not find them. Can you verify that the file `dma.c`, for example, is between your files? It must be inside the `src\main\drivers` folder.
I use the `bash` terminal method in windows 10, so I can help with too much with Cygwin.

stanislavdimov, 2017-04-14:
I use to same. Windows version 10. Yes file which are missin forme are on src\main\drivers folder. Yes some file mybe missing. If is not possible to see all code to understand where are problem.....The project cleanflight is too big to be understand evrything from one people.....Of course I use Cygwin...but this is not my force....my programs I write on C++ in RAD Studio.../last Borland C++/...

Blucena88, 2017-04-15:
McGiverGim It's possible force NAZE32 use the U-BLOX protocol?

McGiverGim, 2017-04-15:
@Blucena88 Yes I've it. But you need to activate the gps compiling your own firmware first.

Blucena88, 2017-04-15:
Ok. I'm starting in programming and i don't know how to create my own firmware. It's possible active this protocol using the current firmware 2.0.0?

McGiverGim, 2017-04-15:
No. You can't activate the gps in naze under this version.

Blucena88, 2017-04-15:
It's is possible install the firmware 1.14.2 and force the NAZE32 to disable option do update?

McGiverGim, 2017-04-15:
Yes, you can install manually any version. The manual versions will not update.

Blucena88, 2017-04-15:
I did that but when i open the cleanflight configurator 2.0 the application request me to update the firmware again. Can I skip or disable this message and open the configuration interface?

McGiverGim, 2017-04-15:
You must install the cleanflight configurator 1.2.4.

Blucena88, 2017-04-15:
Do you have a installer?
I downloaded the file  https://github.com/cleanflight/cleanflight-configurator/tree/v1.2.4 but i don't know how to install a chorme extesion with those files

stanislavdimov, 2017-04-16:
Blucena88 you can not to install CF-configurator version 1 ! Google chrome do not allows. Only ver 2! I write to but no answеr!

McGiverGim, 2017-04-16:
Yes, you can install it: https://blog.hunter.io/install-chrome-extension-manually/

stanislavdimov, 2017-04-16:
Thanks!
But I still  can not manualy compile with GPS! 
Please compile for me /FC - NAZE32 ver6/ 
-with GPS 
- without telemtry,ledstrip,servo,PWM,PPM,spectrum,only 1 serialport,Ibus,  
/I use 1 serialport for GPS and sbas for resiver XSR for TX FrSky Taranis+/!

McGiverGim, 2017-04-16:
It seems that latest master has some problems, so I cannot help you. I recommend to use CF 1.x until master 2.x work again.

stanislavdimov, 2017-04-16:
:)

Blucena88, 2017-04-17:
Thank you for the support McGiverGim and stanislavdimov.
I will try to do this procedure. 
Thank's guys

hydra, 2017-04-17:
if there are confirmed compilation issues please create separate issues with steps to repeat.  the issue tracker is not for user support.  discussions like this are best had in user forums, slack or IRC.

closed 2017-04-17 by hydra