cleanflight/cleanflight #2717: Define of SONAR_TRIGGER_PIN_PWM not used in CF 2.0

S-Klann-DEV, 2017-04-13:
SONAR_TRIGGER_PIN_PWM-Pin is not used in the code. Sonar wont work if someone tries to use it.

sandeeppahadi, 2017-04-14:
i have problem with sonar too. even though i turned sonar feature on its not turning into green. Oneshot125 esc is also not  working. everything was fine before upgrading. Can anyone give me solution?

S-Klann-DEV, 2017-04-17:
To make the sonar working with pwm receiver:

Just define the motor pins M7 and M8 to the SONAR_TRIGGER_PIN and SONAR_ECHO_PIN.

`#define SONAR_TRIGGER_PIN       PB8`
`#define SONAR_ECHO_PIN          PB9`

sandeeppahadi, 2017-04-17:
How to make sonar work on ppm mode?

hydra, 2017-04-18:
Confusing defines removed in 690bedf11ef543b160916203cda042d9671b56dc.  If you need sonar on a naze32 you'll need to compile your own version.

closed 2017-04-18 by hydra