cleanflight/cleanflight #2718: With DSHOT150 no signal on ESC #4

fuxs, 2017-04-13:
I have activated DSHOT150 and now the ESC #4 does not get any signal. ONESHOT150 is still working, thus, it is not a technical issue. Used my oscilloscope and there is absolutely nothing. I am running a SPRacingF3Mini.

emma0223, 2017-04-14:
At least en bf signal output for motor 4 with dshot dosent work since that motor stm32 output can't handle it.. disable the ppm port using bf instructions in the cli

Follow above and to re-map output type in CLI:
resource ppm none
resource motor 4 B04
save 

Then move the signal wire of motor 4 to the ppm output

emma0223, 2017-04-14:
Copied from BF dshot esc protocol instrucciones 

FC Targets that work with Dshot but require hardware mods and re-mapping pins with the Resource command:

Boris' comment:
I had many requests for MOTOLAB (and other FCs) but that one doesn't have DMA available on all motors. It may be that we will assign one of the motors to PPM pin so you can resolder it. Not great, but better than nothing I guess?

General instructions for re-mapping pins:

1- All FCs that require using the FC's PPM input pin as a motor output therefore can NOT use a PPM RX. Any of the Serial RXs that use a UART do work. Set-up Serial RX normally as needed for the FC board.
2- Check pins for the FC board below on which STM32 pins need to be re-mapped. It is a Good Idea to first type in the CLI:
resource
resource list
and copy/paste these into a Text file and save for reference of the Default pin Mappings.
3- In the config tab select OneShot(42 or 125). Click Save. Leave this select until pins are re-mapped.
4- In the CLI type (x = motor #, yyy = STM32 pin #):

resource ppm none    
resource motor x yyy   
save  

SPRACINGF3MINI - Move motor #4 to the PPM pin. Then use resource command to disable PPM and map motor 4 output to B04.

Follow above and to re-map output type in CLI:
resource ppm none
resource motor 4 B04
save

fuxs, 2017-04-14:
Found it this second in the bf docs. Thanks a lot for the quick support!

closed 2017-04-14 by fuxs