cleanflight/cleanflight #2719: Problem in sonar and oneshot125 esc

sandeeppahadi, 2017-04-14:
i have problem with sonar too. even though i turned sonar feature on its not turning into green. Oneshot125 esc is also not working. everything was fine before upgrading. Can anyone give me solution?

McGiverGim, 2017-04-14:
What FC are you using?

sandeeppahadi, 2017-04-14:
i m using spracingF3 delux

McGiverGim, 2017-04-14:
Ok, I've seen that maybe the problem is the same that https://github.com/cleanflight/cleanflight/issues/2717 and that you are aware of that. I think that the only solution is to wait to some developer to solve it and use the 1.x version until then. Let's wait to some developer to see if there's other solution, sorry.

sandeeppahadi, 2017-04-14:
neither 1.x version is working. When I setup previous version It asks to upgade it. I cannot configure with previous version either. I am stucked now. I am using PMW for motors to run cause my motors doesnot run on selecting oneshot125. Everything was fine before upgrading.

McGiverGim, 2017-04-14:
You miss use the older version of the configurator https://github.com/cleanflight/cleanflight-configurator/tree/master_v1.x

hydra, 2017-04-14:
duplicate of https://github.com/cleanflight/cleanflight/issues/2717

closed 2017-04-14 by hydra